vproc/vicuna #1: Simulation fails when memory width is wider than 32 bits without a DCACHE

stevobailey, 2021-09-03:
I changed the `alu/test_configs.conf` file to
```
VREG_W=128  VMEM_W=64   VMUL_W=64
```
and now simulation fails. I believe the data cache is performing a width conversion to reduce a wide Vicuna data port into a 32-bit port for arbitration with the Ibex instruction port. So when you remove the data cache, you cannot have a wide Vicuna data port.

Can you add support for a wide Vicuna data port without a data cache? Or, at the very least, check the input parameters and error when using a wide Vicuna data port without a data cache?

```
[CONFIG ] alu/vxor_8 VREG_W=128  VMEM_W=64   VMUL_W=64
[ ERROR ] alu/vxor_8/vxor_8                    527 cycles (       12 -       539)
incorrect memory content; diff:
...
@@ -1,7 +1,7 @@
 6861651d
-1d191160
+47434b3a
 6a757468
-b21a100b
+e8404a51
 3f44383b
 37424d54
 5e4b5049
make: *** [Makefile:35: alu/vxor_8] Error 1
```

michael-platzer, 2021-09-20:
Hi, sorry for the late reply.

The data cache indeed performs width conversion, and without a data cache, the data port width must match the memory bus width. As a temporary way to increase the data port width without using a data cache, I have added a parameter for the memory bus width. However, when a wider memory bus is used, an instruction cache is required to convert the 32-bit instruction fetch port of Ibex to the bus width.

You could now use a configuration as follows:
```
VREG_W=128  VMEM_W=64   VMUL_W=64  MEM_W=64 ICACHE_SZ=8192
```

I hope this works as a temporary fix while I come up with a more flexible way that allows port width conversion without using caches. Also, please note that the Verilator simulation currently only works with memory bus widths up to 64 bits.

stevobailey, 2021-09-23:
Thanks, I think another acceptable "solution" is to add an assertion, requiring the user to use a DCACHE or keep the bus widths the same.

michael-platzer, 2021-09-24:
Wider memory bus widths are now possible even without an instruction cache, the only requirement is that the memory bus width must match the vector memory interface width if no data cache is used.

I have also added a bunch of assertions to catch invalid parameters.

commit 66a4dc4b (2021-09-24): Allow memory bus widths > 32 bits without instruction cache

Add support for converting the 32 bit wide instruction memory interface
to a wider memory bus without using an instruction cache for the width
conversion.

Fixes #1.
closed 2021-09-24 by michael-platzer