vproc/vicuna #4: Blocking assignments used in always_ff blocks

stevobailey, 2021-10-01:
Please use non-blocking assignments in all always_ff blocks. I see blocking assignments in these locations

```
vproc_alu.sv     lines 364-374
vproc_elem.sv    lines 374-388
vproc_lsu.sv     lines 515-518
vproc_mul.sv     lines 353-356
vproc_sld.sv     lines 324-338
```

michael-platzer, 2021-10-04:
Thanks, this was unintentional and fixing it uncovered a problem in the ELEM unit.

I will add verible (which catches this issue) for linting to the repo and also check for these kind of issues in CI.

closed 2021-10-04 by michael-platzer

stevobailey, 2021-10-04:
Seems like I missed one:
```
vproc_vregfile.sv line 104
```

michael-platzer, 2021-10-05:
Apologies, I missed that one. These issues are now caught by the linting tool that also runs as part of CI.

commit ce0a842f (2021-10-04): Avoid blocking assignments in always_ff blocks

Fixes #4.
commit 73b1c69e (2021-10-05): Avoid one more blocking assignment in always_ff block

Finally fixes #4.