vproc/vicuna #5: Module parameters should have default values

stevobailey, 2021-10-04:
Not sure if this is a Verilog requirement or just good practice, but I'm having issues because not all module parameters have default values. Can you add them? The default can be 0, which may fail gracefully if not overwritten.

```
vproc_alu.sv
vproc_elem.sv
vproc_lsu.sv
vproc_mul_block.sv
vproc_mul.sv
vproc_sld.sv
vproc_vregfile.sv
vproc_vregpack.sv
vproc_vregunpack.sv
```

stevobailey, 2021-10-04:
I found it :)
https://www.accellera.org/images/downloads/standards/v-ams/VAMS-LRM-2-4.pdf
Section 3.4.1 says that, if the parameter is typed, it must have a default value.

michael-platzer, 2021-10-05:
Thanks for researching this! Hopefully I did not miss any parameters, they should all have default values now. Unfortunately, it seems that neither Verible nor Verilator can check this automatically.

commit 91d05c17 (2021-10-05): Assign default values to all parameters

According to the standard, default values shall be specified for all
typed parameters. Fixes #5. No functional changes.
closed 2021-10-05 by michael-platzer