vproc/vicuna #14: Error in makefile in test directory

Mousavikia, 2021-12-30:
Hi I am working on Ibex and I want to attach vicuna as co processor to it... I managed to install rvv branch of gcc but when I run `make alu/vadd_8` I get this error below:
[image]
How can I solve it?

Thank you in advance...

michael-platzer, 2022-01-03:
Which version of Verilator are you using? It seems like it does not support the -G option to override top-level parameter values.

Mousavikia, 2022-01-03:
Hi @michael-platzer  it's 3.874 and I installed it using `sudo apt-get install verilator` what version should I install?

michael-platzer, 2022-01-03:
That version has been released in 2015, please try to upgrade to a more recent version. Based on your screenshot I assume that you are using Ubuntu. [According to pkgs.org](https://pkgs.org/search/?q=verilator), all versions of Ubuntu that are still being supported have moved to more recent versions of Verilator.

The CI pipeline of this repository uses Verilator 4.028 and on my machine I currently use 4.200, which both work fine. Any version number in between should also work.

Mousavikia, 2022-01-03:
Thank you @michael-platzer ... As you mentioned in your earlier comment I upgraded my verilator to the newest version and it is working... I think I will ask more questions about your coprocessor since I have a masters thesis about CNN networks on RiscV processors with Vector extension... so thank you in advance if I ask more questions ahead...

michael-platzer, 2022-01-04:
Glad to hear that simulation works for you now. I am closing this issue, since the problem is resolved.

Please open a new issue if you encounter further problems or have other questions.

closed 2022-01-04 by michael-platzer