vproc/vicuna #17: Test Directory

Mousavikia, 2022-01-10:
Hi @michael-platzer I thought since the title is different I better open new issue and sorry if I am asking a lot but I think probably my low level questions can help you to have great and complete repo as well :)), anyway.... in test directory we can test vicuna and see number of clock cycles in verilator with 2 different configs... I actually want to write the same set of programs (for example vadd_8.s) in C without the help of vector extension just for execution on Ibex alone and see how many clock cycles Ibex will need for them, so that I can compare Ibex with Ibex+Vicuna for a sample vector code. Any suggestions?

michael-platzer, 2022-01-10:
You can run performance tests on the FPGA using [the performance counters of the Ibex core](https://ibex-core.readthedocs.io/en/latest/03_reference/performance_counters.html), which allow you to count the number of elapsed clock cycles.

Scalar equivalents of the vector functions can be implemented in C with for loops. For instance, for the `vadd` instruction, the equivalent C code would be:
```
int i;
for (i = 0; i < vl; i++) {
    vd[i] = vs1[i] + vs2[i];
}
```

Note that for the unit tests in this repository there is a significant overhead (fetching instructions, loading and storing the data, ...) that is also included in the cycle count shown when running the tests. On the other hand, when using performance counters to time just a single vector instruction, you will observe very low values because the vector instructions keep executing in the background. In order to get a reasonable estimate of the cycle count of a vector instruction, you would have to execute that instruction several times, measure the total duration and divide by the number of times the instruction was executed.

Mousavikia, 2022-01-10:
Thank you @michael-platzer plz keep the issue open so that after I read and search about all the great information you mentioned I can ask possible questions about the topic, and again thank you... You are a big help!

closed 2022-06-11 by Mousavikia