vproc/vicuna #19: Scalar core stalling logic is incorrect?

stevobailey, 2022-01-13:
From my understanding, the memory arbiter stalls requests from the scalar core if the vector core has outstanding loads or stores. But I do not think this stalling is correct. Consider the following assembly code:

```
    .text
    .global main
main:
    li              a0, 64
    la              a1, vdata_start
    la              a2, vdata_mid
    la              a3, vref_end

    vsetvli         a4,a0,e32,m8,tu,mu
    vle32.v         v8,(a1)
    vle32.v         v16,(a2)
    vadd.vv         v8,v8,v16
    vse32.v         v8,(a3)

    lw              a4, 0(a3)

    la              a0, vdata_start
    la              a1, vdata_end
    j               spill_cache

    .data
    .align 10
    .global vdata_start
    .global vdata_mid
    .global vdata_end
vdata_start:
    .rept 64
    .word           0x323b3f47
    .endr
vdata_mid:
    .rept 64
    .word           0x47434b3a
    .endr
vdata_end:

    .align 10
    .global vref_start
    .global vref_end
vref_start:
    // not correct, but we don't care for this test
    .rept 64
    .word           0xe2fa599a
    .endr
vref_end:
```

It sets the vector length to 64. It loads 2 vectors, each with 64 32b entries. Then it adds them and stores the result. Then the scalar core immediately tries to read the first word in the result. This read should be stalled until the vector core completes its store instruction. But the waveform shows otherwise.

Higher resolution:
[image]

Arrow points to Ibex memory read, which occurs too early:
[image]

The top group is the data memory interface coming from the scalar core (Ibex). The middle group is the data memory interface coming from the vector core (Vicuna). You can see the memory read from Ibex is granted before Vicuna starts loading its vectors.

I believe this is because `pending_store_o` is not registered, so it is low between the time Vicuna receives the store instruction and the time it actually starts executing it. You can see `pending_store_o` in the bottom group of the above image.

The assembly code above should work in your test framework, though it will fail. But you can run it and generate a waveform to reproduce this issue.

stevobailey, 2022-01-13:
Actually, I don't think the second issue you mentioned is real. As I understand it, here's what is implemented in RTL:

1. If Vicuna has outstanding loads (`pending_load_o` is high), then Ibex can READ from memory, but not WRITE to it.
2. If Vicuna has outstanding stores (`pending_store_o` is high), then Ibex can neither READ nor WRITE to memory.

This looks correct to me.

michael-platzer, 2022-01-26:
This issue should be fixed now. The test case that you provided works as expected, with the scalar load being delayed until the vector loads and stores are complete.

commit efbdcaf0 (2022-01-26): Fix pending load and store signals

Add capability to read flags of entries in a queue to `vproc_queue.sv`
and use this to identify vector load and store instructions currently
held in the instruction queue.  Fixes #19.
closed 2022-01-26 by michael-platzer