vproc/vicuna #20: Help in understing operand widths

stevobailey, 2022-01-20:
Can you help me understand what the operand widths mean? Let's take the default parameters as a starting point:
```
        parameter int unsigned        VREG_W        = 128, // vector register width in bits
        parameter int unsigned        VMUL_W        = 64,  // MUL unit operand width in bits
        parameter int unsigned        VALU_W        = 64,  // ALU unit operand width in bits
        parameter int unsigned        VSLD_W        = 64,  // SLD unit operand width in bits
```
From a RISC-V standards perspective, `VREG_W` equals `VLEN`, so `VLEN` is 128. `ELEN` is 32, since this is a 32-bit vector unit. If `VALU_W` is 64, the operands are 64 bits. If I want to do vector add on 32-bit vectors, can it only add two elements at a time (because 64-bit operands / 32-bit elements = 2 elements per operand)? So doubling `VALU_W` to 128 would require twice the hardware but take half as many cycles to perform the addition?

In your paper, this statement
> The ability to individually configure the throughput for each unit improves the performance of heavily used operations by increasing the respective unit’s data-path width (e.g., widening the data-path of the multiplier unit).

means, if I use lots of multiplies in my application, I should increase `VMUL_W` to improve performance at the cost of HW, right? But if I use lots of adds, I should increase `VALU_W` to improve performance at the cost of HW.

Put another way, `VMUL_W`, `VALU_W`, and `VSLD_W` don't affect functionality (i.e. software), just performance and overhead. Yes?

michael-platzer, 2022-01-21:
Yes, that is correct. The operand width of each unit determines the amount of data that can be processed in one cycle but has no effect on the results. Increasing the operand width of a unit improves performance but requires more HW resources.

However, due to the way Vicuna fetches vector register, the maximum operand width of a unit is half the vector register width, i.e. `VREG_W / 2`. Also, the minimum operand width is 32. Within these limits, the operand width of each unit can be chosen freely (although it must be a power of 2).

stevobailey, 2022-01-21:
Git it, thanks!

closed 2022-01-21 by stevobailey