vproc/vicuna #24: About MakeFile and Linker

Mousavikia, 2022-01-23:
Hi @michael-platzer I have tried to compile 2 simple CNN codes with your Makefile and link.ld file and boot them to Bram of my board using  `bootserdow `. After running `make -f /path/to/vicuna/sw/Makefile PROG=test OBJ=test.o` for the first program I got the following error:
[image]

I have compiled the code for Ibex before using their Makefile but on my other Vmware Ubuntu with following toolchain setup:
`$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain`
`./configure --prefix=/opt/riscv --enable-multilib`
`make`
I know it maybe because of toolchain setups but is there any solution here as well? I am desperately trying to run my written CNN codes here as well and then try to write vector intrinsic of one of them for comparison...
One more thing is after I searched about the error one solution was adding this code:
```
void * memcpy ( void * destination, const void * source, int num ){

  int i=0;

  *((int*)destination) = *((int*)source);

}
```
I did it and the error was gone but when I boot it using `bootserdow `the printed statement was Class is 1 but on Ibex it was class 2 which is correct and C compiler says it is class 2 as well. Is this because of adding the memcpy code? How can I fix it so that first it compiles without any error and second the outputs match?
Here is my simple code:
[test.MD](https://github.com/vproc/vicuna/files/7919849/test.MD)
I had to change the format from .c to .md in order to send it...

Mousavikia, 2022-01-23:
And for the second CNN I have malloc and free functions inside the code as well. Before I used following code at the beginning of all of my C codes that have malloc and free to compile it. Again I have compiled and ran all of this codes on Ibex before. 
```
/* Version of sbrk for no operating system.  */
void*
_sbrk(incr)
int incr;
{
    extern char   end; /* Set by linker.  */
    static char* heap_end;
    char* prev_heap_end;

    if (heap_end == 0)
        heap_end = &end;

    prev_heap_end = heap_end;
    heap_end += incr;

    return (void*)prev_heap_end;
}
```
For this I had to change linker file of Ibex a little bit:
```
OUTPUT_ARCH(riscv)

/* required to correctly link newlib */
GROUP( -lc -lgcc -lsupc++ )

SEARCH_DIR(.)
__DYNAMIC  =  0;

MEMORY
{
    rom         : ORIGIN = 0x00000000, LENGTH = 0x36800 /* 218 kB */
    stack       : ORIGIN = 0x00036800, LENGTH = 0x4F000 /* 316 kB */
}

/* Stack information variables */
_min_stack      = 0x2000;   /* 8K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

/* We have to align each sector to word boundaries as our current s19->slm
 * conversion scripts are not able to handle non-word aligned sections. */

SECTIONS
{
    .vectors :
    {
        . = ALIGN(4);
        KEEP(*(.vectors))
    } > rom

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        *(.text.*)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        . = ALIGN(4);
        _endtext = .;
    }  > rom

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } > rom

    .shbss :
    {
        . = ALIGN(4);
        *(.shbss)
    } > rom

    .data : {
        . = ALIGN(4);
        sdata  =  .;
        _sdata  =  .;
        *(.data);
        *(.data.*)
        edata  =  .;
        _edata  =  .;
    } > rom

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } > rom
    end = .;

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack

    .stab  0 (NOLOAD) :
    {
        [ .stab ]
    }

    .stabstr  0 (NOLOAD) :
    {
        [ .stabstr ]
    }
}
```
Simply I added `end = .;` and the end of `.bss section`. But here for sw directory when I `make -f /path/to/vicuna/sw/Makefile PROG=test OBJ=test.o` first error is still memcpy but even after adding following (which gave diffrent answers in previus comment from original ibex core)
```
void * memcpy ( void * destination, const void * source, int num ){

  int i=0;

  *((int*)destination) = *((int*)source);

}
```
I get `undefined reference to malloc` or `undefined reference to free`... I know it is lots of talking but any suggestions?

stevobailey, 2022-01-23:
You probably need some library code to get those functions. Try adding `-lc -lgcc` to your GCC linker call. Make sure to put these after your .o files. Alternatively, you can try adding `-fno-builtin`.

Mousavikia, 2022-01-23:
Thank you @stevobailey I think in this repo everyone are helping me out... I am really thankful
I change the linker like this:
```
/* Copyright TU Wien                                                          */
/* Licensed under the ISC license, see LICENSE.txt for details                */
/* SPDX-License-Identifier: ISC                                               */

OUTPUT_ARCH(riscv)
GROUP( -lc -lgcc )
ENTRY(_start)

SEARCH_DIR(.)
__DYNAMIC = 0;

MEMORY
{
    boot : ORIGIN = 0x00000000, LENGTH = 0x00002000 /*   8 kB */
    ram  : ORIGIN = 0x00002000, LENGTH = 0x0003E000 /* 248 kB */
}

STACK_LEN = 0x4000; /* 16 kB */

SECTIONS
{
    .vectors 0 : {
        *(.vectors)
    } >boot

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        *(.text.*)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        . = ALIGN(4);
        _endtext = .;
    } >ram

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } >ram

    .shbss : {
        . = ALIGN(4);
        *(.shbss)
    } >ram

    .data : {
        . = ALIGN(4);
        sdata = .;
        _sdata = .;
        *(.data);
        *(.data.*)
        edata = .;
        _edata = .;
    } >ram

    .bss (NOLOAD) : {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } >ram

    .user_align (NOLOAD) : {
        . = ALIGN(0x1000);
        _user_start = .;
    } >ram

    .stack ORIGIN(ram) + LENGTH(ram) - STACK_LEN (NOLOAD) : {
        . = ALIGN(16);
        _stack_start = .;
        . = . + STACK_LEN;
        . = ALIGN(16);
        stack = .;
        _stack = .;
        _stack_end = .;
    } >ram

    .stab 0 (NOLOAD) : {
        [ .stab ]
    }

    .stabstr 0 (NOLOAD) : {
        [ .stabstr ]
    }
}
```
If I understood correctly you mean adding `GROUP( -lc -lgcc )` to the linker right?
But know it gives following error:
[image]
Note: I have build the toolchain with this repos instruction for vector extension

stevobailey, 2022-01-23:
Sorry, that wasn't clear. I mean pass those as arguments to gcc:
```
riscv32-unknown-elf-gcc -march=rv32imv -mabi=ilp32 -static -mcmodel=medany -fvisibilty=hidden -nostdlib -nostartfiles -Wall -T link.ld  crt0.o test.o uart.o -lc -lgcc -o test.elf
```

If that doesn't work, you can pass the `-fno-builtin` argument to all your `riscv32-unknown-elf-gcc` calls.

michael-platzer, 2022-01-25:
The Makefile in [`sw/Makefile`](https://github.com/vproc/vicuna/blob/main/sw/Makefile) does not use the C standard library by default (see the option `-nostdlib` in [lines 17 and 18](https://github.com/vproc/vicuna/blob/e849cd11679ec519396aa1eb7e017b9a23f2e633/sw/Makefile#L17-L18)).

If you want to use C library functions in your code (e.g., `memcpy`, `malloc`, ...), then you can either:
1. Use the C library by removing the `-nostdlib` option or explicitly linking the library by using `-lc -lgcc` as @stevobailey suggested. Note that this requires a C library that is compatible with the target architecture. However, it appears that the C library that you are attempting to use is not compatible. The error `can't link double-float modules with soft-float modules` suggests that the C library uses HW floating point functions. However, neither Ibex nor Vicuna have a floating point unit and thus only support soft-float code (i.e., where all floating point functionality is implemented in SW).
2. Implement the required functions yourself, as you have done for `memcpy`, and/or avoid using library functions.

Note that a bare metal system (that is not managed by an OS) as this one and has no MMU cannot dynamically allocate memory. While it is possible to declare a fixed-size static heap in the linker script, which then allows to use `malloc`, `realloc`, `free` etc., I would suggest using static arrays instead.

closed 2022-06-11 by Mousavikia