vproc/vicuna #26: LMUL is global or per register group?

stevobailey, 2022-01-25:
I'm unsure if the following code is legal or not.

I am attempting to add reduce a vector group down to a single element. I get something like below when compiling with GCC:

```
# initialize v1 to 32-bit element, LMUL=1 value 0
0107f7d7            vsetvli a5,a5,e32,m1,tu,mu
5e0030d7            vmv.v.i v1,0

# now add reduce v4, which is also 32-bit elements but has LMUL=4, into v1
0127f757            vsetvli a4,a5,e32,m4,tu,mu
0240a0d7            vredsum.vs  v1,v4,v1
```
According to the 0.10 spec:
> When LMUL=4, the vector register group contains four vector registers, and instructions specifying an LMUL=4 vector register
group using vector register numbers that are not multiples of four are reserved.

So, since LMUL=4, the source v1 and destination v1 are both flagged by Vicuna as invalid registers, since v1 is not a multiple of 4. But this is a strange case. While LMUL=4 for v4, LMUL=1 for v1. This assumes we can associate different LMUL values for different registers/register groups. So there's no fundamental reason why this instruction is invalid.

I do not know if this is valid or not. Spike correctly simulates this. But LLVM compiles using v8 and v16 instead of v1 and v4, so that avoids this issue. However, LLVM produces the following code elsewhere, which is the same problem. Here, I load an 8-bit vector into v9 using LMUL=1, then load a 16-bit vector into v10 using LMUL=2. Then I sign extend the 8-bit register into 16 bits, from v9 to v12. But LMUL=2 for this instruction, so Vicuna flags v9 as an invalid register.  
```
040576d7            vsetvli a3,a0,e8,m1,ta,mu
02058487            vle8.v  v9,(a1)
049576d7            vsetvli a3,a0,e16,m2,ta,mu
02065507            vle16.v v10,(a2)
4a93a657            vsext.vf2 v12,v9
eec52857            vwmul.vv  v16,v12,v10
```

stevobailey, 2022-01-25:
I think there's a lot more clarity in the 1.0 spec on how to do this. For example, I found the following in the 1.0 spec:
> The scalar input and output operands are held in element 0 of a single vector register, not a vector register group, so any vector register can be the scalar source or destination of a vector reduction regardless of LMUL setting.

So this says the first instruction, `vredsum.vs v1,v4,v1` is valid. And I also found this:
> The vector architecture includes instructions that take multiple source and destination vector operands with different element widths, but the same number of elements. The effective LMUL (EMUL) of each vector operand is determined by the number of registers required to hold the elements. For example, for a widening add operation, such as add 32-bit values to produce 64-bit results, a double-width result requires twice the LMUL of the single-width inputs.

This suggests that the second instruction, `vsext.vf2 v12, v9`, is also valid, since the effective LMUL of the second operand is 1, even if LMUL=2 for the instruction as a whole (i.e. vlmul=2).

michael-platzer, 2022-01-25:
LMUL is a global setting but the effective vector length multiplier (EMUL) is per register group. The instruction `vredsum.vs v1,v4,v1` is legal with an LMUL of 4. Vicuna's decoder incorrectly flags this as illegal.

There seems to be a problem with the reduction operations for LMUL > 1, I will try to address this first and then fix the decoder.

michael-platzer, 2022-01-27:
This issue should be fixed in 9d24761.

closed 2022-01-27 by michael-platzer