vproc/vicuna #35: Linter doesn't like pulling parameters from interfaces

stevobailey, 2022-01-26:
https://github.com/vproc/vicuna/blob/66ed26430c3c99dcd5dbfa5132d2da8a75c145b7/rtl/vproc_core.sv#L69

Unfortunately, some linters and synthesis tools don't accept this syntax. It seems silly, but I usually have to pass in the parameter at the module level (in addition to it begin in the interface).

stevobailey, 2022-01-26:
See #27 and #31 before fixing this issue.

michael-platzer, 2022-01-26:
@stevobailey Thanks, I was not aware that this was a problem with certain tools. Thanks also for your patches!

I am currently in talks with the OpenHW Group who might be interested in evaluating Vicuna in an industrial context. However, they require that Vicuna switches to an Apache-based license (most likely the [Solderpad HW license](https://solderpad.org/)) instead of the current MIT-based ISC license for that.

If that is fine with you, I would merge your pull requests after updating the license (which probably will happen this week). Please let me know if there are any issues with your patches being contributed under a new Apache-based license.

stevobailey, 2022-01-26:
That's really cool! 

My contributions can be under either license.

michael-platzer, 2022-01-27:
Thanks a lot! I have switched to the Solderpad HW license v2.1 and merged your pull requests.

I will add the instruction ID width as a parameter to `vproc_core.sv` to close this issue as well.

commit c1db461b (2022-01-27): Avoid pulling parameters from interfaces

Pass the width of instruction IDs of the XIF interface as a parameter
to `vproc_core.sv`.  This avoids getting this parameter from the
interface, which apparently is not accepted by some tools.  Fixes #35.
closed 2022-01-27 by michael-platzer