vproc/vicuna #40: Adding other peripherals to the demo_top.sv

Mousavikia, 2022-02-10:
Hi @michael-platzer  and @stevobailey  sorry I was unfortunately dealing with covid-19 and couldn't bother you more with my basic questions :p... but here is one:
In demo_top.sv you attached UART, ram32 and hwreg_iface... I want to have Ibex as main core, vicuna as vector co-processor, VGA, camera module and LED's attached as well... I have done this before using wishbone b4 interface with the help of [https://github.com/pbing/ibex_wb](url)... I wonder if I can do it here as well? Even without wishbone interface is it possible to treat those peripherals like URAT and ram32 of your top module code and attach them to ibex memory bus? Any suggestions? (Suppose I will finally attach ddr2 memory of my FPGA board as well)

michael-platzer, 2022-02-17:
Controlling additional peripherals is best implemented with memory-mapped registers, which can be added to [the `hwreg_iface` module](https://github.com/vproc/vicuna/blob/1bccda587118c40a74ebfd71e5895d42ac30df58/demo/rtl/demo_top.sv#L151), similar to how UART is controlled via a status and a data register. In the demo design, these memory-mapped registers occupy the address range from 0x00FF0000 to 0x00FFFFFF, with address 0x00FF0000 currently being used for the UART data and address 0x00FF0004 for the UART status register. The remaining addresses in that range can be used for custom registers.

Mousavikia, 2022-04-18:
Hi @michael-platzer I looked into adding a basic timer with the guide you gave above...
I want to measure runtime of the codes (vector and non vector) codes that I have written to measure the improvement that vicuna cause. I changed `hwreg_iface` like below and tried to do this like uart:
```
module hwreg_iface #(
        parameter int unsigned CLK_FREQ       = 100_000_000,
        parameter int unsigned UART_BAUD_RATE = 115200
    )
    (
        input  logic          clk_i,
        input  logic          rst_ni,
        input  logic          req_i,
        input  logic          we_i,
        input  logic [15:0]   addr_i,
        input  logic [31:0]   wdata_i,
        output logic          rvalid_o,
        output logic [31:0]   rdata_o,
        output logic [15:0]   LED,
        input  logic          rx_i,
        output logic          tx_o
    );

    localparam logic [13:0] ADDR_UART_DATA   = 16'h0000 >> 2;
    localparam logic [13:0] ADDR_UART_STATUS = 16'h0004 >> 2;

    localparam logic [13:0] ADDR_TIMER_DATA   = 16'h0008 >> 2;
    //localparam logic [13:0] ADDR_TIMER_STATUS = 16'h000C >> 2;
    logic       timer_req;
    //logic       timer_wbusy;
    //logic       timer_rvalid;
    logic  [15:0] timer_rdata;
    logic  [15:0] led;

    assign timer_req = req_i && addr_i[15:2] == ADDR_TIMER_DATA;

    timer timer
    (
        .clk_i          ( clk_i             ),
        .rst_ni         ( rst_ni            ),
        .activate       ( timer_req && (wdata_i[31:0] == 32'd1) ),
        .time_o         ( timer_rdata       )
    );

    logic       uart_req;
    logic       uart_wbusy;
    logic       uart_rvalid;
    logic [7:0] uart_rdata;

    assign uart_req = req_i && addr_i[15:2] == ADDR_UART_DATA;

    uart_iface #(
        .CLK_FREQ       ( CLK_FREQ          ),
        .BAUD_RATE      ( UART_BAUD_RATE    )
    ) uart (
        .clk_i          ( clk_i             ),
        .rst_ni         ( rst_ni            ),
        .we_i           ( uart_req && we_i  ),
        .wdata_i        ( wdata_i[7:0]      ),
        .wbusy_o        ( uart_wbusy        ),
        .read_i         ( uart_req && !we_i ),
        .rvalid_o       ( uart_rvalid       ),
        .rdata_o        ( uart_rdata        ),

        .rx_i           ( rx_i              ),
        .tx_o           ( tx_o              )
    );

    logic [31:0] rdata;
    logic        rvalid;

    always_ff @(posedge clk_i) begin
        unique case (addr_i[15:2])
            ADDR_UART_DATA:   rdata <= uart_rvalid ? uart_rdata : 32'hFFFFFFFF;
            ADDR_UART_STATUS: rdata <= { 30'h0, uart_rvalid, uart_wbusy };

            ADDR_TIMER_DATA:   led <= timer_rdata;
            //ADDR_TIMER_DATA:   rdata <= timer_rvalid ? timer_rdata : 32'hFFFFFFFF;
            //ADDR_TIMER_STATUS: rdata <= { 30'h0, timer_rvalid, timer_wbusy };

            default:          rdata <= 0;
        endcase
        rvalid <= req_i;
    end

    assign rdata_o  = rdata;
    assign LED      = led;
    assign rvalid_o = rvalid;
endmodule
```
And for the timer module:
```
module timer(
    input  logic clk_i,
    input  logic rst_ni,
    input  logic activate,
    output logic [15:0] time_o
    );
  logic [15:0] count = 0;
  always @(posedge clk_i or negedge rst_ni) begin
    if (!rst_ni) begin
      count <= 0;
    end else if (activate) begin
      count <= count + 1;
    end
  end
   assign time_o = (count/100000);
endmodule
```
Also in code I activate timer like this:
Beginning of the code:
```
static volatile long *timer_data   = (volatile long *) 0xFF000008;
int main(void) { 
timer_data = 0x00000001;
```
and at the end of the code I disable it like this:
`timer_data = 0x00000000;`
Can you say if the approach and the things that I have done is correct?

Mousavikia, 2022-04-21:
Hi @stevobailey and @michael-platzer ... the above code looks correct for me but the timer doesn't get activated whatsoever... can you plz take a look?

closed 2022-06-11 by Mousavikia