vproc/vicuna #41: Vicuna I$ compliation error when Memory Width and Way Length are same

iksw, 2022-02-11:
I$ configured with size of 4KB, parameter WAY_LEN is 128 bits and if the MEM_BYTE_W is being configured with 128  bit. Then there is a compilation error 
parameter int unsigned MEM_BYTE_W  = 4,  // Memory data width (bytes)
parameter int unsigned WAY_LEN     = 256 // Cache way length (lines)

Compilation Error/data/shared/mulberry/users/kwali/nna_icache_exp/nna/external/vicuna/rtl/vproc_cache.sv, 45
"val"
  Packed union members must have same size.
  Member "val" has different size (1 bits) from next member (3 bits).

michael-platzer, 2022-02-17:
Thanks for pointing this out. The way the cache is currently implemented requires that the line width is larger than the memory data width (i.e., `LINE_BYTE_W` > `MEM_BYTE_W`). I will try to remove this limitation.

In the meantime, please use a larger line width as a workaround, e.g. `ICACHE_LINE_W` == 256.

iksw, 2022-02-17:
I also see that both tag and data memories are implemented as flops with dual port memory interface. Just suggesting if that can be changed to single port memory interface.