vproc/vicuna #45: Aligning memory requests with Ibex

stevobailey, 2022-03-15:
I need Vicuna + Ibex to support unaligned memory requests, but I do not need the predictable timing that Vicuna guarantees. Can you let me know if the following proposal will work? Assume a memory width of 32 bits.

If I pass 0 to the ADDR_ALIGNED parameter, then Vicuna memory requests will output the full address. I will write an adapter that aligns the memory data to the address. So unaligned reads will read two words and shift them down to become aligned. Unaligned writes will write two words by shifting up the write data and byte enable. I can pipeline this to reduce the latency overhead. Do you see an issue with this?

1. Does Vicuna expect all read data to be shifted down? For example, if I read one byte at offset 0x1, will Vicuna expect this to be in the lowest byte of rdata?
2. Does Vicuna expect all reads to be packed? If I read two bytes at offset 0x3, these two bytes span two different words. Will Vicuna expect one valid rdata word with both bytes in the lowest two bytes? Or will it expect two valid rdata words, with each byte in its respective lowest byte?
3. Does Vicuna align write data? If it writes one byte at offset 0x1, will it place this data in the second byte of the word? Or will it keep it in the lowest byte of the wdata word?
4. Does Vicuna pack writes? If it writes two bytes at offset 0x3, will it send two write requests at the two different addresses, or will it send one write request with aligned write data?
5. Same questions as 3 and 4 for the byte enable signal.

stevobailey, 2022-03-15:
Also, how does the answer to all these questions change if we have a strided load/store?

michael-platzer, 2022-03-17:
> If I pass 0 to the ADDR_ALIGNED parameter, then Vicuna memory requests will output the full address. I will write an adapter that aligns the memory data to the address. So unaligned reads will read [...]

If I understand you correctly, you are planning to write an adapter that would align and split accesses in a way similar to how this is currently done by the CV32E40X core. That should work, it should make no difference whether the memory transactions are aligned by the main core or by a dedicated adapter.

> 1. Does Vicuna expect all read data to be shifted down? For example, if I read one byte at offset 0x1, will Vicuna expect this to be in the lowest byte of rdata?

Yes, Vicuna always expects the read data to be shifted into the lowest bits. The byte that should end up in byte 0 of a vector register needs to be the lowest byte of rdata. 

> 2. Does Vicuna expect all reads to be packed? If I read two bytes at offset 0x3, these two bytes span two different words. Will Vicuna expect one valid rdata word with both bytes in the lowest two bytes? Or will it expect two valid rdata words, with each byte in its respective lowest byte?

I assume that you mean reading two bytes at an address with an offset of 0x3 into the lowest bytes of a vector register? In that case Vicuna will only issue a single memory request and expects a single result. The task of splitting the request into two separate transactions and then merging the results into a single rdata word must be done by the adapter.

> 3. Does Vicuna align write data? If it writes one byte at offset 0x1, will it place this data in the second byte of the word? Or will it keep it in the lowest byte of the wdata word?

Vicuna does not align write data for unit-strided loads. When `ADDR_ALIGNED` is 1, then there is some limited alignment for strided and indexed stores that ensures that these execute correctly under the assumption that the base address is aligned to the width of the memory interface. When `ADDR_ALIGNED` is 0, then there is no alignment at all and the write data is always in the lowest bytes of wdata. Thus, one byte written at offset 0x1 will still be kept in the lowest byte of wdata.

> 4. Does Vicuna pack writes? If it writes two bytes at offset 0x3, will it send two write requests at the two different addresses, or will it send one write request with aligned write data?

For unit-strided loads and stores, Vicuna always writes/reads `VMEM_W` bytes at once (irrespective of the current element width), although for writes the byte enable mask signal might prevent writing some of the bytes depending on the current VL and whether a store instruction was masked. Thus, it always performs `VREG_W / VMEM_W` memory requests for loading/storing one vector register, with the address incremented by `VMEM_W / 8` each time.

For strided or indexed loads and stores, Vicuna processes each element individually and generates one request for each element. In that situation the number of requests depends on the current element width setting. When `ADDR_ALIGNED` is 0 each element is always shifted into the lowest bytes of wdata for stores and is expected to be in the lowest bytes of rdata for loads.

> 5. Same questions as 3 and 4 for the byte enable signal.

The byte enable signal is handled in the same way as the write data in wdata. The n-th bit of the byte enable mask always corresponds to the n-th byte of wdata.

I hope that my answers help clarify the requirements for your adapter. Please let me know if anything is unclear or if you have any further questions!

stevobailey, 2022-03-21:
Thanks, this makes sense and the adapter appears to be working.

closed 2022-03-21 by stevobailey