vproc/vicuna #48: Issue executing vloxei8.v

moimfeld, 2022-03-21:
Hi @michael-platzer

## Issue
I am mostly done with setting up the simulation environment for Questasim. I can run most test cases successfully except the vloxei8.v instructions.

The issue arises when the `vproc_tb` is in the following configuration:

	MEM_W=32
	MEM_SZ=262144
	MEM_LATENCY=5
	VREG_W=512
	VMEM_W=256
	VMUL_W=128
	VALU_W=128
	VSLD_W=128
	ICACHE_SZ=8192
	ICACHE_LINE_W=128
	DCACHE_SZ=65536
	DCACHE_LINE_W=512

	*this is the second configuration defined in the .config files in the
	test directories (and the default values set in the sim Makefile)

while simulating the execution of the [vloxei8.S](https://github.com/vproc/vicuna/blob/main/test/lsu/vloxei8_8.S) code (Main core is ibex). The issue is that the coprocessor (and ibex) gets stuck.

## Findings

- The coprocessor completes five out of eight instruction (I concluded this because the ready/valid handshake of the result interface happens five times until the coprocessor gets stuck)
- This means the coprocessor gets stuck during the following instruction: `vloxei8.v       v16, (a1), v8, v0.t`
- During the execution of this particular instruction the valid signal of the memory request/response channel stays high for 23 clock cylces. For the first 16 clock cylces a "valid" address is assigned to the `mem_req.addr` signal. Afterwards, the `mem_req.addr` signal goes to an undefined state (e.g. `32'hXXXXXXX0`)
- This "invalid" address causes unwanted behavior in the cache module (i.e. the `cpu_gnt_o` signal goes to `1'bX`)
- In total, the handshake of the memory request/response channel is completed (both valid and ready are high) 19 times during the execution of this instruction. Four times valid is high while ready is in an undefined state.

I have not further investigated this issue, but the problem is either:

- the assignment of the valid signal of the memory request/response channel
- the address calculation/assignment
	- if the issue is in the address calculation, then there could also be an issue with the operands or the register file (e.g. my asic register file)

## Question

Before I continue investigating, I would like to ask: Is this a known issue? If not, then I might have introduced it with my ASIC vector register or it is caused by using Questasim.

michael-platzer, 2022-03-22:
Hi @moimfeld , thanks for reporting this issue and the detailed analysis!

This is not an issue that I am aware of. The address of the XIF memory request should be valid anytime `mem_valid` is asserted.

The `vloxei8.v` instruction generates the memory address by adding indices read from a vector register to a base address. Maybe some of the indices in that vector register are invalid or there is an issue with the address generation logic.

Unfortunately, I am not able to reproduce the issue with either Verilator or Vivado. Can you identify the source of these don't care values?

stevobailey, 2022-03-22:
I ran into a similar issue with other instructions. Because the compute modules are fractured, the upper and lower register bits are packed as a single input to the operator. If you try to do, say, an 8-bit add using a vector length of 1, `X`s in the upper bits can pollute valid data in the lower bits. Functionally, there's nothing wrong, since after the operation only the lower bits are written back to memory.

I agree with @michael-platzer, it should not be necessary to initialize the register file to 0. It's not as serious of an issue, but if you're looking into fixing the `vloxei8.v` issue, maybe you could look into suppressing issues with partial use of fractured operators?

As a side note, do these memory reads to invalid addresses set `be` to 0 in the memory interface? You could just suppress any memory transaction that has `be` of 0, but this kills your timing predictability, I think?

moimfeld, 2022-03-23:
Hi @michael-platzer and @stevobailey

Here is a screenshot of the memory channels when the register file is not initialized to 0 on reset. As you can see, the `be` signal also goes to don't care, as soon as the addresses are don't care.

[image]

michael-platzer, 2022-03-23:
@stevobailey Did you have issues with simulation tools that incorrectly set the lower bits to don't care if some higher bits are invalid? I encountered such an issue before, but thought that these do no longer occur. I will look into it.

@moimfeld Thanks for checking the `be` signal, it puzzles me why this goes to don't care as well, it should not be affected by the address. I will try to figure out why that happens.

moimfeld, 2022-03-23:
@michael-platzer I have already spent some time analyzing the signals of the memory channel (also the `be` signal). This is why I think the following lines let the `be` signal go to don't care.
https://github.com/vproc/vicuna/blob/51bd47f9f1f746e6b4a924b5c0805e99d20a76e5/rtl/vproc_lsu.sv#L256-L260
`wmask_buf_d` becomes `wmask_buf_q` and then the signal is directly assigned to `mem_req.be`. `wmask_buf_d` goes to don't care because it is shifted by a don't care value of `req_addr_d` (see screenshot). So the `be` signal is indirectly influenced by the address.

[image]

michael-platzer, 2022-03-23:
@moimfeld Thanks a lot for analyzing this!

Now I see why this happens. The `be` signal and the data are partially aligned for strided and indexed loads and stores and the alignment is based on the address, which is invalid... so even though the mask is 0, the shift turns it into a don't care.

stevobailey, 2022-03-23:
Yes, I had issues. Now I convert any X values into random bits to get around the issue.

moimfeld, 2022-07-04:
Since the test I created a while ago completes successfully, I close this issue.

closed 2022-07-04 by moimfeld