vproc/vicuna #53: Intrinsic Code on Vicuna

Mousavikia, 2022-03-30:
Hi @michael-platzer  and @stevobailey I hope you are doing well... I have been working on RiscV V ISA and its intrinsic and I want to start coding in intrinsic level on Vicuna ..
I have written this simple vector add code:

```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(int* ref, int* actual, int n) {
	for (int i = 0; i < n; ++i)
		if (ref[i] - actual[i] == 0) {
			return 0;
		}
	return 1;
}

// index arithmetic
void add(int* a, int* b, int* c, int n) {
	for (int i = 0; i < n; ++i) {
		a[i] = b[i] + c[i];
	}
}

void vec_add(int* a, int* b, int* c, int n) {
	while (n > 0) {
		size_t vl = vsetvl_e32m1(n);
		vint32m1_t vb = vle32_v_i32m1(b, vl);
		vint32m1_t vc = vle32_v_i32m1(c, vl);
		vint32m1_t va = vadd_vv_i32m1(vb, vc, vl);
		vse32_v_i32m1(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}

}

int main(void) {
	// data
	int a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	int b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };

	// compute
	int result[31];
	int vecresult[31];
	add(result, a, b, 31);
	vec_add(vecresult, a, b, 31);

	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}

```
All I want to do is check whether or not normal vector add and vector add with V extension give same answer or not...
I compile the code it gives warnings about pointers in vector add function but it doesn't throw any error... but when I implement it on the board the uart doesn't give any output... not even "fail"...
Do you know the problem?

michael-platzer, 2022-04-04:
Hi @Mousavikia , the code of your `vec_add` function looks correct and should do what you expect. If you do not get any output at all, then you might want to try narrowing down the problem by starting to check whether printing to UART works right at the start of your program (which should then be akin to a simple "Hello world" program) and then walk your way through your code and see at which point it fails.

Mousavikia, 2022-04-04:
Hi @michael-platzer thank you for your response... I actually tested what you said and following results happened which is making me commit suicide :(( ....
I tested this code that you gave me couple of months ago:
```
#include <uart.h>

int main(void) {
    uart_puts("Hello world!\n");
    return 0;
}
```
This works fine... I even tested a complete CNN and it works and prints right class number via uart...
About this code:
I tested this (added simple hello world right at the beginning of the main function) :
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(int* ref, int* actual, int n) {
	int r;
	for (int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

// index arithmetic
void add(int* a, int* b, int* c, int n) {
	for (int i = 0; i < n; ++i) {
		a[i] = b[i] + c[i];
	}
}

void vec_add(int* a, int* b, int* c, int n) {
	while (n > 0) {
		size_t vl = vsetvl_e32m1(n);
		vint32m1_t vb = vle32_v_i32m1(b, vl);
		vint32m1_t vc = vle32_v_i32m1(c, vl);
		vint32m1_t va = vadd_vv_i32m1(vb, vc, vl);
		vse32_v_i32m1(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}

}
int main(void) {
	uart_puts("Hello world!\n");
	// data
	int a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	int b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };

	// compute
	int result[31];
	int vecresult[31];
	add(result, a, b, 31);
	vec_add(vecresult, a, b, 31);

	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}
```
It doesn't work **and even when I comment codes of vector mode it again doesn't work** but as soon as I delete the vector code like below suddenly it works:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(int* ref, int* actual, int n) {
	int r;
	for (int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

// index arithmetic
void add(int* a, int* b, int* c, int n) {
	for (int i = 0; i < n; ++i) {
		a[i] = b[i] + c[i];
	}
}

int main(void) {
	uart_puts("Hello world!\n");
	// data
	int a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	int b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };

	// compute
	int result[31];
	int vecresult[31];
	add(result, a, b, 31);
	//vec_add(vecresult, a, b, 31);

	// compare
	//uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}
```
I am just out of my mind right now.... please suggest...

Mousavikia, 2022-04-11:
Hi @michael-platzer and @stevobailey... If you really have a clue regarding my issue please help...
I attached the  .vmem file in text (since github doesn't support .vmem format) of the vector add code that I mentioned in first comment... the linker file is the default version and the compiler is gcc...
[vadd.txt](https://github.com/vproc/vicuna/files/8463815/vadd.txt)

stevobailey, 2022-04-11:
Have you tried simulating it? Otherwise, you can add ILAs to probe what's going on inside the FPGA.

Mousavikia, 2022-04-13:
Hi @stevobailey I tested it in vivado with the testbench inside of the demo folder but I completely get lost inside all those signals... I wonder if that .vmem file that I have send it to you works for you when you run it or not... the wired part is all codes without vector functions works fine for me but like above vector add example when I have vector code it doesn't give me any output

Mousavikia, 2022-04-15:
Hi @stevobailey and @michael-platzer I could finally implement the code and get the correct response out of uart... this time I just used .vmem and copy and paste it directly to the .mem file(not utilizing .elf with the bootloader project)

closed 2022-04-23 by Mousavikia