vproc/vicuna #54: LLVM Installation Error

Mousavikia, 2022-04-03:
Hi @moimfeld I have noticed that you added llvm installation in `sw` directory... So, I tried to make llvm with `make llvm LLVM_DIR=/opt/llvm` but first I got cmake version error (I had old version of cmake but llvm installation required newer version)... After I updated my cmake version I tried again and this time I get this error:
```
HEAD is now at 5177676... Updated MLIR type stubs to work with pytype
/bin/sh: 5: cmake: not found
/home/kiian/Desktop/VicunaRepos/vicuna/sw//toolchain.mk:22: recipe for target 'llvm' failed
make: *** [llvm] Error 127
```
It's wired because I have cmake and when I run `cmake --version` terminal responds with this:
```
cmake version 3.22.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
I have ubuntu 16.04 on vmware. Can you help me solve this problem? I tried more than couple of times and I can't solve it

moimfeld, 2022-04-03:
Hi @Mousavikia

You can try to change the value of the `CMAKE` variable in the `toolchain.mk` file to the path of the newer version of cmake (so you would have to change it from `cmake` to something like this: `/path_to_cmake_install/bin/cmake`).

Mousavikia, 2022-04-03:
Thank you for the quick reply... I did what you say:
```
llvm: check_env_vars
	cd $(LLVM_DIR) && mkdir install &&                                     \
	git clone --recurse-submodules $(LLVM_REPO) && cd llvm-project &&      \
	git checkout $(LLVM_COMMIT) && rm build/ -r -f &&                      \
	mkdir build && cd build &&                                             \
	$(/usr/bin/cmake) -G "Unix Makefiles"                                           \
	         -DCMAKE_C_COMPILER=$(CC)                                      \
	         -DCMAKE_INSTALL_PREFIX=$(LLVM_INSTALL_DIR)                    \
	         -DLLVM_ENABLE_PROJECTS=$(LLVM_ENABLE_PROJECTS)                \
	         -DCMAKE_BUILD_TYPE=$(MAKE_BUILD_TYPE)                         \
	         -DLLVM_DEFAULT_TARGET_TRIPLE=$(LLVM_DEFAULT_TARGET_TRIPLE)    \
	         -DLLVM_TARGETS_TO_BUILD=$(LLVM_TARGETS_TO_BUILD)              \
	         ../llvm && cd .. &&                                           \
	$(/usr/bin/cmake) --build build --target install && cd $(LLVM_DIR) &&           \
	git clone $(NEWLIB_REPO) newlib && cd newlib &&                        \
	git checkout $(NEWLIB_COMMIT) &&                                       \
	rm -rf build && mkdir -p build && cd build &&                          \
	../configure --prefix=${LLVM_INSTALL_DIR}                              \
	--target=riscv32-unknown-elf                                           \
	CC_FOR_TARGET="${LLVM_INSTALL_DIR}/bin/clang -march=rv32gc -mabi=ilp32 \
	-mno-relax -mcmodel=medany"                                            \
	AS_FOR_TARGET=${LLVM_INSTALL_DIR}/bin/llvm-as                          \
	AR_FOR_TARGET=${LLVM_INSTALL_DIR}/bin/llvm-ar                          \
	LD_FOR_TARGET=${LLVM_INSTALL_DIR}/bin/llvm-ld                          \
	RANLIB_FOR_TARGET=${LLVM_INSTALL_DIR}/bin/llvm-ranlib &&               \
	make &&                                                                \
	make install
```
hopefully I understood correctly, but now it throws following error :((
```
HEAD is now at 5177676... Updated MLIR type stubs to work with pytype
/bin/sh: 5: -G: not found
/home/kiian/Desktop/VicunaRepos/vicuna/sw//toolchain.mk:22: recipe for target 'llvm' failed
make: *** [llvm] Error 127
```

moimfeld, 2022-04-03:
Sorry that I was not clear enough but you only have to change line 5 (see below). Instead of `cmake` write the path after the `:=`. I think in your case you have to replace `cmake` with `/usr/bin/cmake` (And revert any other changes that you made to the Makefile).

https://github.com/vproc/vicuna/blob/eb18ab3029ed7684c14f40e6c0ae2111e4bc8faf/sw/toolchain.mk#L5

Mousavikia, 2022-04-04:
Thank you for making it more clear...
I did what you said but now another error :((
Looks like it doesn't want to get installed after all:
```
[ 84%] Building CXX object tools/lld/ELF/CMakeFiles/lldELF.dir/SyntheticSections.cpp.o
/opt/llvm/llvm-project/lld/ELF/SyntheticSections.cpp: In member function ‘void lld::elf::RelocationBaseSection::partitionRels()’:
/opt/llvm/llvm-project/lld/ELF/SyntheticSections.cpp:1654:17: warning: variable ‘relativeRel’ set but not used [-Wunused-but-set-variable]
   const RelType relativeRel = target->relativeRel;
                 ^
/opt/llvm/llvm-project/lld/ELF/SyntheticSections.cpp: In instantiation of ‘lld::elf::RelocationBaseSection::partitionRels()::<lambda(auto:3&)> [with auto:3 = lld::elf::DynamicReloc]’:
/usr/include/c++/5/bits/stl_algo.h:1518:21:   required from ‘_BidirectionalIterator std::__partition(_BidirectionalIterator, _BidirectionalIterator, _Predicate, std::bidirectional_iterator_tag) [with _BidirectionalIterator = lld::elf::DynamicReloc*; _Predicate = lld::elf::RelocationBaseSection::partitionRels()::<lambda(auto:3&)>]’
/usr/include/c++/5/bits/stl_algo.h:4520:30:   required from ‘_BIter std::partition(_BIter, _BIter, _Predicate) [with _BIter = lld::elf::DynamicReloc*; _Predicate = lld::elf::RelocationBaseSection::partitionRels()::<lambda(auto:3&)>]’
/opt/llvm/llvm-project/llvm/include/llvm/ADT/STLExtras.h:1696:24:   required from ‘auto llvm::partition(R&&, UnaryPredicate) [with R = llvm::SmallVector<lld::elf::DynamicReloc, 0u>&; UnaryPredicate = lld::elf::RelocationBaseSection::partitionRels()::<lambda(auto:3&)>]’
/opt/llvm/llvm-project/lld/ELF/SyntheticSections.cpp:1656:77:   required from here
/opt/llvm/llvm-project/lld/ELF/SyntheticSections.cpp:1656:60: internal compiler error: in tsubst_copy, at cp/pt.c:13189
       llvm::partition(relocs, [=](auto &r) { return r.type == relativeRel; }) -
                                                            ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
tools/lld/ELF/CMakeFiles/lldELF.dir/build.make:551: recipe for target 'tools/lld/ELF/CMakeFiles/lldELF.dir/SyntheticSections.cpp.o' failed
make[3]: *** [tools/lld/ELF/CMakeFiles/lldELF.dir/SyntheticSections.cpp.o] Error 1
make[3]: Leaving directory '/opt/llvm/llvm-project/build'
CMakeFiles/Makefile2:71338: recipe for target 'tools/lld/ELF/CMakeFiles/lldELF.dir/all' failed
make[2]: *** [tools/lld/ELF/CMakeFiles/lldELF.dir/all] Error 2
make[2]: Leaving directory '/opt/llvm/llvm-project/build'
Makefile:155: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/opt/llvm/llvm-project/build'
/home/kiian/Desktop/VicunaRepos/vicuna/sw//toolchain.mk:22: recipe for target 'llvm' failed
make: *** [llvm] Error 2
```
I wonder what can I do right now...
By the way since you are answering very rapidly, I have opened another issue [https://github.com/vproc/vicuna/issues/53](https://github.com/vproc/vicuna/issues/53) about a vector add intrinsic code that I want to run on vicuna can you look at that and see if you know the problem?
I really appreciate your help

Mousavikia, 2022-04-04:
Ok thanks... I checked the version as you said:
```
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
So it should have been ok! :((

Mousavikia, 2022-04-06:
Hi @moimfeld I hope I haven't bothered you much...
As you said the error was because of gcc version... Although I had version 5.4 which is above 5.1 again it looks like it wasn't enough... So I updated gcc using [this](https://askubuntu.com/questions/781972/how-can-i-update-gcc-5-3-to-6-1) now I have version 6.5, and then I rebuild the whole cumbersome process again :(( ... This time it looks like it has been installed successfully:
[image]
But here is problem when I am compiling my C codes using llvm:
I changed the makefile in `sw` directory and I passed llvm as the default compiler... rest of the file is unchanged...
`COMPILER ?= gcc` to `COMPILER ?= llvm` it is in line 11
But when I compile I get this error:
```
kiian@ubuntu:~/Desktop/VicunaRepos/vicuna/sw$ make
clang -march=rv32imzve32x -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -fuse-ld=lld  -c -o /home/kiian/Desktop/VicunaRepos/vicuna/sw//crt0.o /home/kiian/Desktop/VicunaRepos/vicuna/sw//crt0.S
clang: warning: argument unused during compilation: '-mabi=ilp32'
clang: warning: argument unused during compilation: '-nostartfiles'
clang: warning: argument unused during compilation: '-fuse-ld=lld'
error: invalid value 'medany' in '-mcode-model medany'
Makefile:52: recipe for target '/home/kiian/Desktop/VicunaRepos/vicuna/sw//crt0.o' failed
make: *** [/home/kiian/Desktop/VicunaRepos/vicuna/sw//crt0.o] Error 1
```
Any ideas?

moimfeld, 2022-04-06:
Hi @Mousavikia,

Does this issue only arise if you try to compile C code? Or does it also arise if you just try to compile and run the assembly tests?

Mousavikia, 2022-04-06:
Thanks for the quick reply @moimfeld .... I tested what you said and unfortunately yes the error exists for assembly code as well... I tested this using `test` directory by running:
`make alu/vadd_8`
I changed the compiler to llvm like this:
```
# select the compiler to use; either gcc (default) or llvm
COMPILER ?= llvm
```
Error:
```
[ ERROR ] alu/vadd_8.vmem build failed; content of /home/kiian/Desktop/VicunaRepos/vicuna/test/alu/vadd_8_build.log:
make[1]: Entering directory '/home/kiian/Desktop/VicunaRepos/vicuna/test/alu'
clang -march=rv32imzve32x -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -fuse-ld=lld  -c -o /home/kiian/Desktop/VicunaRepos/vicuna/test/alu/vadd_8.o /home/kiian/Desktop/VicunaRepos/vicuna/test/alu/vadd_8.S
clang: warning: argument unused during compilation: '-mabi=ilp32'
clang: warning: argument unused during compilation: '-nostartfiles'
clang: warning: argument unused during compilation: '-fuse-ld=lld'
error: invalid value 'medany' in '-mcode-model medany'
/home/kiian/Desktop/VicunaRepos/vicuna/test//../sw/Makefile:52: recipe for target '/home/kiian/Desktop/VicunaRepos/vicuna/test/alu/vadd_8.o' failed
make[1]: *** [/home/kiian/Desktop/VicunaRepos/vicuna/test/alu/vadd_8.o] Error 1
make[1]: Leaving directory '/home/kiian/Desktop/VicunaRepos/vicuna/test/alu'
Makefile:153: recipe for target 'alu/vadd_8.vmem' failed
make: *** [alu/vadd_8.vmem] Error 1
```
When it is on gcc it works fine...

moimfeld, 2022-04-06:
It's strange because we seem to have the same compiler now, but it works on my side and not on yours. Can you run `clang --version` and copy what it says?

As a side note: You do not need to change the COMPILER variable in the Makefile. You can also just run: `make alu/vadd8 COMPILER=llvm`. This will overwrite the COMPILER variable from the Makefile. This works because the COMPILER variable is set using `?=` in the Makefile.

Mousavikia, 2022-04-06:
Ok let me copy and paste it:
```
kiian@ubuntu:~/Desktop/vicuna/sw$ clang --version
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```
Thanks for the side note....

moimfeld, 2022-04-06:
Ok, so it looks like when you call `clang` it runs the preinstalled compiler by default. As you can see from `clang --version` the target is you host machine and not risc-v.

Now there are two options that could fix the problem:

1. Possible solution (if this does not work try solution 2):
Run the following command in the terminal: `set PATH=/opt/llvm/install/bin:$PATH`

2. Possible solution:
In the sw/Makefile change the value (I mean the value behind the `:=`) of the `RISCV_CC`, `RISCV_DUMP` and `RISCV_OBCP` variables by prepending this path `/opt/llvm/install/bin`. (So you only have to change the lines you see below)
so for example in line 19 change `clang` to `/opt/llvm/install/bin/clang`

https://github.com/vproc/vicuna/blob/eb18ab3029ed7684c14f40e6c0ae2111e4bc8faf/sw/Makefile#L19-L21

**Important Note:** I assumed that `/opt/llvm/install/bin` is the path where the LLVM + Clang binaries are located on your computer. If this is not the case you have to adapt the path. To check if `/opt/llvm/install/bin` is the correct path run `/opt/llvm/install/bin/clang --version` in the terminal.

Mousavikia, 2022-04-06:
Thank you a lot @moimfeld ... Second solution worked for me and I finally compiled my codes via llvm...
I did all this because I wanted to check that vadd code that I mentioned in [here](https://github.com/vproc/vicuna/issues/53) works when I compile it with llvm or not but it doesn't work with neither gcc nor llvm so the problem is in some where else... I really hope @michael-platzer can help in that regard since I am out of options...
Anyway thank you @moimfeld

closed 2022-04-15 by Mousavikia