vproc/vicuna #55: Reduction Sum Intrinsic

Mousavikia, 2022-04-17:
Hi @michael-platzer , @stevobailey  and @moimfeld ... I have a question about reduction sum intrinsic function:
I have written following code in normal C:
```
signed short int dense(signed char* inDense, signed char* wf, signed short int inDenseSize, signed short int biaseDense) {
	signed short int i;
	signed short int outDense = 0;
	for (i = 0; i < inDenseSize; i++) {
		outDense += inDense[i] * wf[i];
	}
	outDense += biaseDense;
	return outDense;
}
```
I want to convert it into vector mode... I have written following for this purpose but it seems to have a problem since I cannot execute it on vicuna.
```
signed short int vec_dense(signed char* in1, signed char* in2, signed short int n, signed short int bias) {
	signed short int out;
	size_t vlmax = vsetvlmax_e16m1();
	vint16m1_t vec_zero = vmv_v_x_i16m1(0, vlmax);
	vint16m2_t vout = vmv_v_x_i16m2(0, vlmax);
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vout = vwmul_vv_i16m2(vin1, vin2, vl);
		in1 += vl;
		in2 += vl;
		n -= vl;
	}
	vint16m1_t vec_sum = vredsum_vs_i16m2_i16m1(vec_zero, vout, vec_zero, vlmax);
	out = vmv_x_s_i16m1_i16(vec_sum);
	return out + bias;
}
```
 Can you suggest?

stevobailey, 2022-04-19:
This works for me. Which CPU are you using? When you say "it seems to have a problem since I cannot execute it on vicuna", what problem are you seeing?

Mousavikia, 2022-04-20:
Hi @stevobailey  thanks for the reply... cpu is Ibex (I always use vicuna with Ibex) and about the problem: the `dense` function works but when I use `vec_dense` I cannot see anything on uart... complete version of the code is like this:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

signed short int dense(signed char* inDense, signed char* wf, signed short int inDenseSize, signed short int biaseDense) {
	signed short int i;
	signed short int outDense = 0;
	for (i = 0; i < inDenseSize; i++) {
		outDense += inDense[i] * wf[i];
	}
	outDense += biaseDense;
	return outDense;
}

signed short int vec_dense(signed char* in1, signed char* in2, signed short int n, signed short int bias) {
	signed short int out;
	size_t vlmax = vsetvlmax_e16m1();
	vint16m1_t vec_zero = vmv_v_x_i16m1(0, vlmax);
	vint16m2_t vout = vmv_v_x_i16m2(0, vlmax);
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vout = vwmul_vv_i16m2(vin1, vin2, vl);
		in1 += vl;
		in2 += vl;
		n -= vl;
	}
	vint16m1_t vec_sum = vredsum_vs_i16m2_i16m1(vec_zero, vout, vec_zero, vlmax);
	out = vmv_x_s_i16m1_i16(vec_sum);
	return out + bias;
}

int main(void) {
	// data
	signed char a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	signed char b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,-22,23,24,25,-26,-27,28,29,30,-31 };
	signed short int bias = 1;
	// compute
	signed short int result;
	signed short int vecresult;
	result = dense(a, b, 31, bias);
	vecresult = vec_dense(a, b, 31, bias);
	// compare
	uart_puts((vecresult == result) ? "pass" : "fail");
	return 0;
}
```
I should note that the following version of `vec_dense` with just widening multiplication works:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(signed short int* ref, signed short int* actual, signed short int n) {
	signed short int r;
	for (signed short int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

void vec_mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vint16m2_t vout = vwmul_vv_i16m2(vin1, vin2, vl);
		vse16_v_i16m2(out, vout, vl);
		in1 += vl;
		in2 += vl;
		out += vl;
		n -= vl;
	}

}

void mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	signed short int i;
	for (i = 0; i < n; i++) {
		out[i] = in1[i] * in2[i];
	}
}

int main(void) {
	// data
	signed char a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	signed char b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,-22,23,24,25,-26,-27,28,29,30,-31 };
	// compute
	signed short int result[31];
	signed short int vecresult[31];
	mul(result, a, b, 31);
	vec_mul(vecresult, a, b, 31);
	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
	return 0;
}
```

stevobailey, 2022-05-06:
I asked because Ibex does not support memory requests NOT aligned to the word size (32 bits). Here are a few things to try. The first two are related to this memory request alignment issue.
1. Try switching everything to 32 bits and see if the vector functions work.
2. Try changing your array size to be a multiple of 4. Because your data are 8 bits, this should guarantee that your array addresses are aligned to the word boundary.
3. This is a lot more work, but I recommend you set up Spike ISA simulation. This will help you distinguish between issues in your C code and issues in the RTL/design. 
4. If none of the above help you diagnose the issue, you'll probably have to simulate the RTL and inspect the waveform.

Also, which compiler are you using? GCC?

Mousavikia, 2022-05-07:
Hi @stevobailey I utilized GCC as the compiler... Actually, I thought all of the reduction functions doesn't work but then I ran this code:
```
void vec_maxPool(signed char* in, signed char* out, signed short int outSize) {
	signed char n = 4;
	signed char j = 0;
	size_t vlmax = vsetvlmax_e8m1();
	vint8m1_t vec_zero = vmv_v_x_i8m1(0, vlmax);
	vint8m1_t vin1 = vmv_v_x_i8m1(0, vlmax);
	while (outSize != 0) {
		while (n > 0) {
			size_t vl = vsetvl_e8m1(n);
			vin1 = vle8_v_i8m1(in, vl);
			in += vl;
			n  -= vl;
		}
		vint8m1_t vec_max = vredmax_vs_i8m1_i8m1(vec_zero, vin1, vec_zero, vlmax);
		vint8m1_t vin1 = vmv_v_x_i8m1(0, vlmax);
		out[j] = vmv_x_s_i8m1_i8(vec_max);
		outSize -= 1;
		n = 4;
		j++;
	}
}
```
This has reduction max and it works...
I think `redmax `has a problem... I don't know if it's because that non-aligned addresses you mentioned or sth else. Anyway, I will try your suggestions and hopefully I can get the answer since I am running out of time...
One more thing: You said this works for me in your previous comment... You meant that you implemented my code on vicuna+ibex and uart responded "pass" to you? Or you meant that code looks fine but you didn't implement it?
Beside, I utilized this config for vicuna for all of my codes:
```
    vproc_top #(
        .VREG_W       ( 128                         ),
        .VMEM_W       ( 32                          ),
        .VMUL_W       ( 64                          ),
        .VALU_W       ( 64                          ),
        .VSLD_W       ( 64                          ),
        .RAM_TYPE     ( vproc_pkg::RAM_XLNX_RAM32M  ),
        .MUL_TYPE     ( vproc_pkg::MUL_XLNX_DSP48E1 )
    ) vproc (
        .clk_i        ( clk                         ),
        .rst_ni       ( rst_n                       ),
        .mem_req_o    ( mem_req                     ),
        .mem_addr_o   ( mem_addr                    ),
        .mem_we_o     ( mem_we                      ),
        .mem_be_o     ( mem_be                      ),
        .mem_wdata_o  ( mem_wdata                   ),
        .mem_rvalid_i ( mem_rvalid                  ),
        .mem_err_i    ( mem_err                     ),
        .mem_rdata_i  ( mem_rdata                   ),
        .pend_vreg_wr_map_o (                       )
    );
```
I send it because I want to know if I am doing sth wrong... This config was the only thing that I changed from your code since my FPGA is small and I couldn't implement the config you mentioned in your code
I really hope either you (@stevobailey ) or @michael-platzer answer....

closed 2022-06-11 by Mousavikia