vproc/vicuna #58: [Question] Out-of-order result transaction

moimfeld, 2022-04-28:
Hi @michael-platzer,

I have a few questions regarding an observation I made:

### Observation
While simulating vicuna I found that some instruction sequences (for example [vle8_8.S](https://github.com/vproc/vicuna/blob/main/test/lsu/vle8_8.S)) have out-of-order (OoO) result transactions. I know that this is allowed by the x-interface, but I am not sure whether this is intended on vicuna.

The figure below shows the waveforms of the x-interface while executing vle8_8.S. It can be seen that the third offloaded vector instruction (`vmv.v.v2, v0`) with `id` = 2 has its result transaction before the second offloaded vector instruction (`vle8.v v0, (a0)`) with `id` = 1. The same is true for the fourth and fifth instructions.

[image]

### How to replicate
The following command and configuration was used (and the simulation completed successfully):
```
make lsu/vle8_8 SIMULATOR=questa COMPILER=llvm
[CONFIG ] lsu/vle8_8 VREG_W=128  VMEM_W=32   VMUL_W=32
[SUCCESS] lsu/vle8_8/vle8_8                    293 cycles (       12 -       305)
```
Additionally the ram_type was set to RAM_ASIC.

### Questions

1. Is this behaviour expected (I mean OoO result transactions)?
2. If this is not expected, can you replicate the observation stated above (just to be sure that the issue was not introduced by my asic vector register)?
3. I thought `vle8.v  v0, (a0)` followed by `vmv.v.v v2, v0` would cause a data hazard (i.e. hindering the move instruction from executing before the load instruction has completed), why does this not happen here?

moimfeld, 2022-04-28:
@michael-platzer Thank you very much for this explanation! It is very clear now what happens.

This information is very important for me, because in order to verify Vicuna I need to extract the state of the vector register and the vector CSRs after each completed instruction. Such that Vicuna's state can be compared to a golden model to verify correct execution.

Now since the result transaction does not coincide with the actual completion of the instruction I must find a signal (or a few signals) with which I can identify all actual instruction completions.

I have found the following signals that can identify actual instruction completion. 

- `xif_result_if.result_valid`: for loads, stores, configuration and move instructions.
- `vregfile_wr_en_q`: for the remaining instructions

#### Follow-up question
Since you designed Vicuna I want to ask you this follow up question: Are the aforementioned signals sufficient to indicate **all** CSR and vector register state changes caused by instruction completion? Do you know if there exists another signal which could be better suited to identify state change?
(If this is not so easy to answer, I can also figure it out myself)

### Sidenotes

- I am aware that if I use `vregfile_wr_en_q` as a signal to identify a state change, I must wait one cycle before extracting the vector state (because only then the vector register will be updated).
- I am aware, that `xif_result_if.result_valid` will create "false indications" because it is asserted also for empty results which do not cause a state change (this is not a big an issue)
- I am aware, that `vregfile_wr_en_q` will create "false indications" because it is asserted also for memory instruction writebacks (this is not a big an issue)
- I am aware, that if I extract the state of vicuna when `vregfile_wr_en_q` is asserted the configuration  might have already changed (i.e. it might not be the same one used for the instruction that caused the vreg writeback).

michael-platzer, 2022-04-29:
> Now since the result transaction does not coincide with the actual completion of the instruction I must find a signal (or a few signals) with which I can identify all actual instruction [...]

For configuration instructions the XIF result transaction is indeed triggered in the same cycle as the changes are applied to the corresponding register (i.e., signals `vsew_q`, `lmul_q`, `agnostic_q`, `vl_0_q`, `vl_q`, `vl_csr_q`, `vstart_q`, `vxrm_q`, and `vxsat_q` in `vproc_core.sv`). However, due to the different result sources the result transaction may get delayed if it collides with another result (the file `vproc_result.sv` contains the arbitration logic for the various types of results).

For loads and stores the result transaction is triggered as soon as all memory transactions are complete. Yet, writeback to the register file may still be incomplete at that moment.

The following signals should allow to track the actual completion of instructions more reliably (all these signals are part of the `vproc_core` module in `vproc_core.sv`):

* `result_csr_valid` for configuration changes (CSR reads/writes and vsetvl), with the signal `result_csr_id` containing the ID of the associated instruction. This signal actually triggers the register update and the new value is available from the corresponding configuration register (`vsew_q`, `lmul_q`, ...) in the following cycle.
* `instr_complete_valid` for all other instructions. Note that this signal is a vector, since each unit can independently signal completion of an instruction. `instr_complete_id` is the corresponding vector of instruction IDs, so if `instr_complete_valid[i]` is asserted, then `instr_complete_id[i]` contains the associated instruction ID. A unit asserts its `instr_complete_valid` signal in the same cycle as the last writeback to the vector register file occurs. Thus, the updated data is available from the vector register file in the subsequent cycle.

I hope this information helps and that I have not overlooked anything. Please let me know if you encounter any issues when tracking the state of Vicuna.

I have some small changes of Vicuna planned (partially due to the updates of the XIF spec in openhwgroup/core-v-xif#52). I would like to coordinate these with you to ensure that I am not interfering with your work. I will send you an e-mail regarding a potential meeting to discuss this.

moimfeld, 2022-04-29:
Hi @michael-platzer,
Thanks a lot again for your answer. Your help is super useful! With these signals I can track the vector register and vector CSR state without having to filter out any false positives.

I am confident that these signals will be sufficient to track Vicunas vector register and CSR state. In any case, my UVM environment will throw errors if some offloaded (and commited) instruction does not have any associated tracked state change.

I will close this issue now and answer your email.

closed 2022-04-29 by moimfeld