vproc/vicuna #60: Coprocessor stalls indefinitely if memory result transaction happens in the same clock cycle as memory request transaction

moimfeld, 2022-05-04:
Hi @michael-platzer

## Issue
This issue is related to #59, but for delay on a different signal. The problematic delay here is `mem_result_valid`. When a memory request is served in the same clock cycle as it was issued (i.e. no delay), the coprocessor will stall forever. Below you can find a picture of the x-interface signals. After 490 ns the coprocessor stalls indefinitely. I have not further investigated this observation.

This is the corresponding snippet from the [x-interface documentation](https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/en/latest/x_ext.html#interface-dependencies) which states that memory result transactions can happen in the same cycle as the memory request transaction.
> Memory result interface transactions cannot be initiated before the corresponding memory request interface handshake is completed. They are allowed to be initiated at the same time as or after completion of the memory request interface handshake.

[image]

### Problematic Instruction Sequence
The problematic instruction sequence is the same as in #59.

michael-platzer, 2022-06-06:
Hi @moimfeld,

I believe this issue should now be resolved in f17acb3. However, I get a syntax error in `cvxif_vseq_base.svh` when attempting to run the corresponding test in your UVM testbench. Please, let me know if the changes do indeed fix the issue.

moimfeld, 2022-06-08:
Hi @michael-platzer,
Sorry for the delayed answer, but f17acb3 does indeed resolve this bug. Thanks a lot!
I will close this issue now.

closed 2022-06-08 by moimfeld