vproc/vicuna #62: Coprocessor stalls indefinitely if core issues random memory result transactions

moimfeld, 2022-05-04:
Hi @michael-platzer,

## Issue
The x-interface specification state:

> Note that a coprocessor shall be able to tolerate memory result transactions for which it did not perform the corresponding memory request handshake itself.

**It might be beneficial to understand/ask what the reason is behind this rule is before addressing this issue.** If you don't know the reason then I can open a issue on the x-interface repository.

My UVM environment can be configured to generate random memory result transaction (i.e. asserting `mem_result_valid` with unrelated `id` and random `data` / `exc` / `dbg` signals). At the moment there is a problem with generating "unrelated" `id` because the memory request transaction does not have a defined `id` (see #61).

Still, what I find is that the coprocessor stalls indefinitely when it is "bombarded" with random unrelated memory transactions. You can see the coprocessor stall in the screenshot below.

[image]

### Problematic Instruction Sequence
The problematic Instruction Sequence is the same as in #59

michael-platzer, 2022-05-06:
Hi @moimfeld,

thanks for reporting. I believe the reasoning behind tolerating unrelated memory responses is that the XIF interface should be able to connect multiple coprocessor to a main core. Each coprocessor might then generate its own memory requests and should ignore any memory results that are destined for other coprocessors (by comparing the instruction ID of the memory result transaction with the ID used in its request).

As an initial step towards resolving this issue I have modified Vicuna's LSU such that it ignores memory result transactions unless it actually expects a result.

I also tried to ignore the result transaction if the ID does not match the expected value, but that currently leads to problems which I do no yet understand and will have to investigate further.

moimfeld, 2022-05-06:
Hi @michael-platzer,

Indeed, this rule makes sense in a multi-core environment.

Your fix resolved the problem of "infinite" stalls. Thanks a lot!
But as you mentioned, the coprocessor should probably also check if the `id` corresponds to the one of its request, or else it might store/accept wrong results.

michael-platzer, 2022-06-08:
Hi @moimfeld,

this issue should now be fixed in 60ec4a7. Memory result transactions for which the instruction ID does not match the ID specified during the associated request are now ignored. Please let me know if re-enabling random memory results in your UVM environment causes issues.

moimfeld, 2022-06-09:
Hi @michael-platzer,

The infinite stalls are gone 😄. Thanks a lot!

closed 2022-06-09 by moimfeld