vproc/vicuna #63: RAM_ASIC register file has multiply driven registers

stevobailey, 2022-05-11:
https://github.com/vproc/vicuna/blob/38500cab0109c0d4afe51c2519ef0e85fe5b29b2/rtl/vproc_vregfile.sv#L161

If there are two write ports, and both try to write to the same address on the same cycle, how is this conflict resolved? The current implementation requires multiple drivers for each register and is not implementable on an ASIC.

If you can guarantee there will never be write conflicts, or you don't care what data are written when a conflict occurs, you can OR all the write inputs together like this:
``` systemverilog
    logic [VREG_W-1:0]   ram[32];
    logic [VREG_W-1:0]   ram_next[32];
    logic [VREG_W/8-1:0] ram_wbe[32];

    // OR the inputs from each port
    always_comb begin
      for (int i = 0; i < 32; i ++) begin
        ram_next[i] = '0;
        ram_wbe[i] = '0;
        for (int j = 0; j < PORTS_WR; j ++) begin
          if (wr_we_i[j] && (wr_addr_i[j] == i)) begin
            for (int k = 0; k < PORT_W/8; k++) begin
              if (wr_be_i[j][k]) begin
                ram_next[i][k*8 +: 8] |= wr_data_i[j][k*8 +: 8];
                ram_wbe[i][k] |= 1;
              end
            end
          end
        end 
      end
    end

    // write
    always_ff @(posedge clk_i) begin
        for (int i = 0; i < 32; i++) begin
            for (int k = 0; k < PORT_W/8; k++) begin
                if (ram_wbe[i][k]) begin
                  ram[i][k*8 +: 8] <= ram_next[i][k*8 +: 8];
                end
            end
        end
    end

    // read
    always_comb begin
        for (int i = 0; i < PORTS_RD; i++) begin
            rd_data_o[i] = ram[rd_addr_i[i]];
        end
    end
```

Otherwise, you'll need some kind of arbitration or duplication to handle this.

michael-platzer, 2022-05-12:
Multiple write ports should never write to the same register in the same cycle. The dispatcher (implemented in `vproc_dispatcher.sv`) maintains a record of outstanding vreg writes and stalls instructions that would write to the same register until a pending write is complete.

I believe that it should be safe to OR the inputs together, as you suggest.

@moimfeld Since you contributed the ASIC register file, what is your opinion on this?

moimfeld, 2022-05-12:
Hi @stevobailey and @michael-platzer

This is indeed a very good point. When I implemented the ASIC register file I did not think about this case. And when compiling the code using QuestaSim, it did not throw an error. Fortunately, Michael can guarantee that two writes to the same address will not occur at the same time.

So I agree with you that we should OR the inputs (I can fix the issue and open a PR).

Additionally, I can write an assertion that checks if the addresses of the write ports are never the same, while `we` is high for both ports.