vproc/vicuna #69: Syntax error in vproc_config generation

kuoyaoming93, 2022-06-02:
Hi @michael-platzer,

I see that there is syntax error in vproc_config package. Please look into config.mk

Some reports screenshoots from Vivado:
[image]

[image]

michael-platzer, 2022-06-02:
Hi @kuoyaoming93,

unfortunately I cannot reproduce this. Which shell are you using? Maybe the double square brackets are the problem, I have replaced them by single square brackets (that apparently are more portable).

Please let me know if that fixes your issue.

michael-platzer, 2022-06-03:
> It seems that the script add an aditional "," when the first parameter is empty.

It should be the other way around, the comma should only be added if `pipe_units` is *not* empty. The error message in the second screenshot is definitely related, line 45 of the shell script corresponds to [line 103 of `config.mk`](https://github.com/vproc/vicuna/blob/main/config.mk#L103), which is the conditional statement that decides whether a comma should be inserted or not. Apparently your shell thinks that the operator `[` should not be here and as a consequence the `if`-statement is always false, thus taking the `else`-branch and prepending commas to the parameter value strings:

https://github.com/vproc/vicuna/blob/b0c2aa2a1791ca2277a5890e37b138ecec64870f/config.mk#L103-L115

What is your `/bin/sh` pointing at, i.e. could you paste the output of `ls -l /bin/sh` and maybe also of `/bin/sh --version`?

kuoyaoming93, 2022-06-03:
Hi. There are my outputs:

[image]

[image]

kuoyaoming93, 2022-06-03:
Hi @michael-platzer, it is correctly generated using **bash**, not **dash**.
If there is a solution to automatically select bash as default could be great. I proceed to close this issue. 

[image]

```verilog
package vproc_config;

    import vproc_pkg::*;

    parameter vreg_type    VREG_TYPE                   = VREG_GENERIC;
    parameter int unsigned VREG_W                      = 128;
    parameter int unsigned VPORT_RD_CNT                = 2;
    parameter int unsigned VPORT_RD_W   [VPORT_RD_CNT] = '{default: VREG_W};
    parameter int unsigned VPORT_WR_CNT                = 1;
    parameter int unsigned VPORT_WR_W   [VPORT_WR_CNT] = '{default: VREG_W};

    parameter int unsigned PIPE_CNT                    = 1;
    parameter bit [UNIT_CNT-1:0] PIPE_UNITS [PIPE_CNT] = '{
        (UNIT_CNT'(1) << UNIT_LSU) | (UNIT_CNT'(1) << UNIT_ALU) | (UNIT_CNT'(1) << UNIT_MUL) | (UNIT_CNT'(1) << UNIT_SLD) | (UNIT_CNT'(1) << UNIT_ELEM)
    };
    parameter int unsigned PIPE_W           [PIPE_CNT] = '{32};
    parameter int unsigned PIPE_VPORT_CNT   [PIPE_CNT] = '{1};
    parameter int unsigned PIPE_VPORT_IDX   [PIPE_CNT] = '{1};
    parameter int unsigned PIPE_VPORT_WR    [PIPE_CNT] = '{0};

    parameter int unsigned VLSU_QUEUE_SZ               = 4;
    parameter bit [VLSU_FLAGS_W-1:0] VLSU_FLAGS        = '0;
    parameter mul_type     MUL_TYPE                    = MUL_GENERIC;

    parameter int unsigned INSTR_QUEUE_SZ              = 2;
    parameter bit [BUF_FLAGS_W-1:0] BUF_FLAGS          = '0;

endpackage

```

closed 2022-06-03 by kuoyaoming93

michael-platzer, 2022-06-03:
The Makefiles should now work with `dash` (and hopefully most other shells) as well.

kuoyaoming93, 2022-06-03:
Thank you for all @michael-platzer! 

Confirmed here that work now with dash, which is the default sh in Ubuntu.

commit 8511bc0f (2022-06-03): Fix conditional statements in shell code

Fix several conditional statements in shell code that caused issues with
other shells than bash.  Fixes #69.