vproc/vicuna #71: Measuring run time of C codes in normal and vector mode

Mousavikia, 2022-06-03:
Hi... I have added a timer and led module to demo project and then I tried to measure run time of a vector add code like below and compare the run time with the normal version of the code(without vector instructions):
Vector processor config:
```
    vproc_top #(
        .VREG_W       ( 128                         ),
        .VMEM_W       ( 32                          ),
        .VMUL_W       ( 64                          ),
        .VALU_W       ( 64                          ),
        .VSLD_W       ( 64                          ),
        .RAM_TYPE     ( vproc_pkg::RAM_XLNX_RAM32M  ),
        .MUL_TYPE     ( vproc_pkg::MUL_XLNX_DSP48E1 )
    ) vproc (
        .clk_i        ( clk                         ),
        .rst_ni       ( rst_n                       ),
        .mem_req_o    ( mem_req                     ),
        .mem_addr_o   ( mem_addr                    ),
        .mem_we_o     ( mem_we                      ),
        .mem_be_o     ( mem_be                      ),
        .mem_wdata_o  ( mem_wdata                   ),
        .mem_rvalid_i ( mem_rvalid                  ),
        .mem_err_i    ( mem_err                     ),
        .mem_rdata_i  ( mem_rdata                   ),
        .pend_vreg_wr_map_o (                       )
    );
```
Normal C code:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}
void add(signed short int* a, signed short int* b, signed short int* c, signed short int n) {
	for (int i = 0; i < n; ++i) {
		a[i] = b[i] + c[i];
	}
}
static volatile long* timer_data = (volatile long*)0xFF000008;
int main(void) {
	//start
	*timer_data = 0x00000001;
	// data
	signed short int x[1024] = { 19,5,73,35,67,39,63,3,92,81,75,82,39,62,58,54,28,25,46,23,81,99,3,54,9,81,99,7,94,2,69,79,54,89,90,63,14,22,19,5,11,62,94,36,42,99,95,68,99,77,34,67,25,30,69,53,42,61,76,59,56,59,52,9,72,100,36,98,35,89,46,42,22,13,31,73,79,70,1,85,93,78,5,38,71,73,23,27,68,48,63,24,18,83,77,94,11,19,10,49,20,90,10,5,56,78,32,18,34,22,52,91,63,11,40,6,51,44,100,82,49,90,14,40,93,92,72,62,35,94,13,74,65,84,40,75,84,33,56,98,55,34,62,37,76,42,50,70,98,33,84,74,96,4,36,67,29,24,72,63,60,67,5,35,46,25,72,86,29,74,14,84,14,59,37,81,51,49,88,36,45,97,5,98,19,67,59,68,37,63,82,2,9,98,66,24,41,13,27,26,34,16,35,13,89,10,94,40,5,35,74,80,55,69,90,6,31,5,20,73,73,88,59,8,93,81,29,55,99,72,84,44,48,57,27,75,51,65,31,14,48,37,79,79,67,14,3,56,31,94,99,29,81,90,60,89,95,55,73,58,3,45,65,53,38,94,83,85,38,60,88,94,67,21,66,8,41,67,94,82,49,76,42,98,99,87,39,46,25,79,89,92,56,60,15,90,46,21,90,77,89,29,68,67,13,41,28,72,29,90,83,40,50,70,84,61,58,33,46,72,89,73,2,68,44,44,12,82,33,25,35,38,55,57,40,40,52,66,96,73,41,84,14,7,9,17,33,31,2,54,10,15,64,86,98,58,100,56,52,34,44,50,8,89,7,44,83,40,62,82,89,94,20,26,90,60,51,62,82,54,21,46,43,97,63,70,73,35,52,56,16,57,70,43,84,74,37,46,39,78,74,44,70,95,79,71,11,39,60,46,6,23,84,2,87,8,67,51,22,58,13,68,60,6,6,16,2,44,84,62,53,87,10,91,11,52,15,56,1,77,85,92,99,51,28,11,51,59,77,9,67,52,18,94,60,45,95,66,46,84,54,56,69,37,24,58,87,41,12,45,31,41,84,41,40,37,15,27,9,43,26,30,43,12,50,71,25,79,8,40,1,23,1,19,15,27,18,14,60,91,94,23,49,38,53,27,7,44,18,3,96,44,97,77,1,69,71,65,56,22,78,23,38,90,86,41,32,61,92,91,60,34,86,45,91,4,54,72,18,34,19,33,41,55,5,56,28,25,25,16,96,94,82,73,18,37,19,1,32,70,63,55,44,29,51,77,77,58,75,65,13,51,35,10,15,20,68,44,70,26,1,54,28,95,91,40,3,68,84,98,6,46,59,69,72,66,73,38,59,12,6,98,29,60,97,19,20,35,94,40,28,16,40,38,14,44,10,62,2,58,79,24,45,57,7,50,65,23,84,98,85,51,28,75,24,96,63,61,18,10,26,86,92,70,73,23,58,82,41,99,9,33,52,7,73,56,53,83,86,79,32,46,76,11,11,27,53,98,72,32,30,86,92,64,26,9,84,59,95,7,59,29,83,20,45,40,83,68,21,32,14,68,58,17,15,48,91,56,4,6,81,46,39,79,37,54,72,88,33,66,98,8,59,42,31,27,76,100,19,79,20,100,81,43,73,50,81,36,8,60,92,20,44,75,4,95,77,56,19,50,52,100,86,97,68,41,94,48,24,40,71,56,76,100,97,54,97,12,6,31,59,54,91,55,44,55,72,2,81,15,48,26,37,67,17,28,20,20,33,89,48,41,18,97,41,85,62,38,88,79,47,82,90,43,34,60,91,71,38,74,96,55,55,32,8,19,10,47,1,92,65,1,4,21,46,13,1,73,36,79,44,44,5,5,10,60,25,85,86,97,49,23,23,54,77,35,47,64,92,17,72,58,44,89,40,18,64,63,33,81,100,99,13,24,3,61,12,41,89,55,37,21,45,96,13,48,86,5,70,98,29,14,69,91,62,90,20,76,35,42,16,82,63,74,81,7,96,50,76,75,84,16,46,62,94,84,90,59,59,86,4,89,41,4,75,16,15,61,26,33,41,41,39,61,17,19,10,33,77,24,75,70,83,83,30,31,53,33,84,82,56,27,69,24,46,39,54,100,76,99,24,53,6,76,61,86,99,93,41,1,55,21,22,33,10,75,75,55,34,84,56,96,90,36,55,35,63,80,75,13,83,3,42,74,79,37,75,90,25,13,23,36,29,93,6,60,17,84,17,51,100,36,5,22,40,34,23,94,69,97,44,95,1,62,81,24,94,77,83,58,80,33,23,32,59,83,30,41,87,62,100,21,83,68 };
	signed short int y[1024] = { 25,48,40,60,81,11,83,85,36,44,58,71,75,76,39,43,96,58,85,28,63,59,97,9,51,53,10,91,89,44,79,15,62,27,45,85,20,31,49,34,80,99,16,24,71,38,98,98,65,87,41,64,99,56,94,73,49,64,89,20,40,100,41,66,91,100,66,11,4,62,57,97,75,67,53,26,97,55,4,70,52,6,90,34,23,12,32,23,66,7,28,29,89,45,76,61,79,12,98,85,6,47,33,64,24,58,61,60,45,4,52,41,11,46,46,56,81,71,88,6,22,46,96,80,46,34,6,75,51,20,43,17,76,37,95,2,83,63,54,66,73,10,88,2,30,18,93,7,59,64,66,87,6,82,53,70,22,55,71,96,45,9,6,63,80,70,35,95,53,96,8,21,78,92,79,30,16,85,79,28,23,33,83,83,58,58,29,70,80,45,45,47,28,68,91,91,75,27,69,14,13,20,15,59,8,83,73,93,50,66,90,54,29,98,4,33,98,37,31,13,92,14,34,90,50,62,59,70,3,53,4,83,34,85,25,59,94,5,6,3,69,60,12,80,62,8,7,14,79,10,24,25,11,86,70,74,66,52,33,67,12,15,2,97,98,13,47,66,30,76,56,43,27,76,90,73,41,94,26,54,96,27,26,93,7,30,60,21,64,80,51,66,80,24,61,12,52,84,93,50,28,66,92,51,98,20,12,30,40,43,32,70,10,41,30,31,11,60,29,16,1,29,56,88,5,91,14,84,81,92,14,51,41,18,58,61,22,52,99,49,70,42,4,30,81,35,9,52,37,74,53,81,82,19,13,83,64,2,90,52,55,61,77,86,39,9,74,34,84,38,83,18,13,88,5,69,74,44,38,98,40,45,16,33,32,90,25,32,41,71,15,88,9,47,4,76,71,22,68,56,86,56,91,42,36,49,26,93,47,26,44,71,41,19,86,59,38,23,22,53,44,75,8,85,68,14,86,20,61,55,17,1,78,77,43,6,59,18,73,54,26,92,76,89,7,19,74,70,78,51,43,62,86,68,53,30,71,39,57,89,85,90,94,82,1,1,9,27,3,43,35,55,93,30,34,86,35,14,51,86,39,70,63,46,48,95,9,28,45,59,88,47,44,75,47,87,47,50,49,23,9,7,89,24,87,72,88,94,14,40,99,65,90,49,2,63,24,53,73,61,59,44,25,43,2,61,96,10,4,89,25,1,82,15,88,10,36,60,59,67,65,44,14,76,25,66,86,9,98,4,84,84,5,55,95,33,81,61,79,80,5,29,66,49,98,75,57,30,26,89,45,82,10,86,3,90,90,53,13,18,71,84,4,76,96,35,64,35,22,79,73,28,59,43,10,3,50,28,34,29,18,40,70,21,67,45,44,18,20,62,27,56,95,72,68,96,78,61,95,6,27,99,78,48,69,42,39,22,39,3,48,34,98,56,85,41,47,83,100,53,93,74,57,97,83,96,65,38,48,92,2,16,48,55,6,66,89,11,44,29,99,61,26,14,55,83,84,84,21,55,88,13,86,90,22,8,48,84,47,42,51,13,14,88,61,27,87,6,46,73,34,41,53,90,78,7,28,38,87,42,24,60,48,90,94,82,71,75,90,7,34,1,83,51,37,23,54,29,7,9,7,41,13,45,90,36,13,57,88,35,5,15,8,75,46,67,70,58,63,88,67,88,47,15,7,72,31,68,66,54,72,51,49,50,94,39,12,25,69,84,98,22,77,59,41,52,50,66,75,31,9,83,39,78,18,11,91,88,100,87,4,55,100,52,88,8,99,93,57,44,34,73,2,38,93,55,48,50,31,96,99,52,100,46,43,22,20,84,73,53,83,52,56,22,59,15,6,69,61,22,41,63,56,13,17,1,42,49,16,67,2,12,96,98,4,50,87,25,84,82,63,1,38,91,69,38,64,25,58,99,85,29,69,36,99,9,26,82,9,54,81,74,15,44,36,48,59,15,91,65,17,57,94,79,69,47,27,57,25,32,92,89,80,93,18,52,63,92,67,39,75,82,61,9,93,6,53,12,39,82,25,89,72,38,25,26,77,5,69,63,75,98,39,27,88,81,47,10,57,19,54,36,32,73,52,80,21,68,6,81,68,95,10,91,51,62,32,8,86,15,38,63,100,52,100,23,40,70,7,75,43,82,38,32,99,72,42,10,74,64,8,13,99,50,3,6,15,90,47,57,50,7,90,29,27,60,48,37,66,94,63,29,21,44,3,88,62,21,52,6,87,45,55,57,69,38,8,46,5,74,4,96,75,94,52,25,26,76,100,36,76,12,60,44,74,27,10,46,65,14,46,66,83,31 };

	// compute
	signed short int result[1024];
	add(result, x, y, 1024);

	// stop
	*timer_data = 0x00000000;
	while (1);
}
```
Vector mode C code:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
[... 57 lines cut ...]
        .mem_rdata_i  ( mem_rdata                   ),
        .pend_vreg_wr_map_o (                       )
    );
```
I compiled the vector mode code again using llvm but the delay is again 2ms which is really weird because it should become less.
[... cut ...]
void vec_add(signed short int* a, signed short int* b, signed short int* c, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e16m8(n);
		vint16m8_t vb = vle16_v_i16m8(b, vl);
		vint16m8_t vc = vle16_v_i16m8(c, vl);
		vint16m8_t va = vadd_vv_i16m8(vb, vc, vl);
		vse16_v_i16m8(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}
}
```
Can someone explain what am I missing?
I send the demo_top.sv code so that you can see how I added the timer and led module to the core... Am I doing sth wrong? I am confused...
[Code.zip](https://github.com/vproc/vicuna/files/8832012/Code.zip)

kuoyaoming93, 2022-06-03:
Hi @Mousavikia,

If you are using an FPGA, maybe you can evaluate directly the number of cycles reading RISC-V built-in CSR.
https://github.com/vproc/vicuna/blob/main/sw/lib/runtime.c

Example C code:
```c
#include <runtime.h>
#include <uart.h>

int main(void){
    long time1,time2;
    time1 = get_mcycle();
    asm volatile("vle16.v v0, (%0);" ::"r"(&wires[0][0])); 
    asm volatile("vle16.v v2, (%0);" ::"r"(&wires[1][0])); 
    asm volatile("vadd.vv v4, v0, v2;");
    asm volatile("vse16.v v4, (%0);" ::"r"(&result[0]));
    time2 = get_mcycle();

    uart_printf("The operation took %d cycles\n\r",time2-time1);
    while(1);
    return 0;
}
```

[image]

michael-platzer, 2022-06-03:
Hi @Mousavikia,

without having looked at your timer implementation, there are a couple of things to consider when trying to compare scalar and vector code:
- The peak performance of Vicuna is determined by the size of the operands, not the size of the vector registers. Increasing the vector register width from 128 to 256 bits as you did has little impact on performance if you do not also increase the width of the operands (i.e., the parameters `VALU_W`, `VMUL_W`, `VSLD_W`; note that since merging #43 these have been replaced with a more generic way of configuring Vicuna).
- Your code has 3x more vector loads/stores than arithmetic instructions, which means that it has a very low arithmetic intensity and is therefore memory-bound (i.e., the overall performance will primarily depend on the vector load and store instructions). The performance of vector loads/stores is limited by the width of the memory interface (`VMEM_W`), which in your example is 32 bits. Therefore, vector loads/stores have the same memory bandwidth available and thus cannot perform faster than the loads and stores of your non-vector code example.
- Vector instructions tend to have very long latency. This means that if an instruction depends on the result of a prior vector instruction, it might have to wait a while for that result to become available, before it can start executing. For instance, the `vadd` instruction in your code needs the results of the two preceding `vle` instructions, and then the `vse` needs to wait for the result of `vadd` before being able to proceed. Arranging vector instructions in a way that avoids data dependencies between consecutive instructions helps avoid stalls and keep the execution units busy. You might want to have a look at the assembly code that your compiler is generating to see how it arranges these vector instructions. Performance might significantly improve if you load a bunch of data into multiple vector registers first and then perform the `vadd`. I am working on a scheduling model of Vicuna for the LLVM compiler, which will provide the necessary information to LLVM to automatically re-arrange (schedule) vector instructions in a way that optimizes performance.
- Your code is processing an array with 1024 16-bit elements, which means that with a vector register width of 256 bits and a vector multiplier setting of 8 you will have a `VL` of 128 and thus only 8 loop iterations. That is not terribly much, considering that there might be some overhead involved in starting/stopping the timer and calling the `vec_add` function (depending on the level of optimization that your compiler uses, of course). Maybe you might want to increase the length of the array to compensate for any potential overhead.

Mousavikia, 2022-06-03:
Wow, you worked a lot when I was gone :) ... Thank you both @michael-platzer and @kuoyaoming93 for your answers...
I will discuss your kind answers with my advisor to get the whole picture hopefully... But one more observation that I had in the past two days:
I compiled that vector add code with both llvm and gcc... 
The gcc was installed like below I think 6 months ago with your repos guide but llvm installation was newer when you introduced how to install llvm in this repo:
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic
cd riscv-gnu-toolchain
git submodule update --init --recursive
mkdir build && cd build
../configure --prefix=/desired/installation/path
make
```
The code again:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

void vec_add(signed short int* a, signed short int* b, signed short int* c, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e16m1(n);
		vint16m1_t vb = vle16_v_i16m1(b, vl);
		vint16m1_t vc = vle16_v_i16m1(c, vl);
		vint16m1_t va = vadd_vv_i16m1(vb, vc, vl);
		vse16_v_i16m1(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}

}
static volatile long* timer_data = (volatile long*)0xFF000008;
int main(void) {
	//start
	*timer_data = 0x00000001;
	// data
	signed short int x[1024] = { 19,5,73,35,67,39,63,3,92,81,75,82,39,62,58,54,28,25,46,23,81,99,3,54,9,81,99,7,94,2,69,79,54,89,90,63,14,22,19,5,11,62,94,36,42,99,95,68,99,77,34,67,25,30,69,53,42,61,76,59,56,59,52,9,72,100,36,98,35,89,46,42,22,13,31,73,79,70,1,85,93,78,5,38,71,73,23,27,68,48,63,24,18,83,77,94,11,19,10,49,20,90,10,5,56,78,32,18,34,22,52,91,63,11,40,6,51,44,100,82,49,90,14,40,93,92,72,62,35,94,13,74,65,84,40,75,84,33,56,98,55,34,62,37,76,42,50,70,98,33,84,74,96,4,36,67,29,24,72,63,60,67,5,35,46,25,72,86,29,74,14,84,14,59,37,81,51,49,88,36,45,97,5,98,19,67,59,68,37,63,82,2,9,98,66,24,41,13,27,26,34,16,35,13,89,10,94,40,5,35,74,80,55,69,90,6,31,5,20,73,73,88,59,8,93,81,29,55,99,72,84,44,48,57,27,75,51,65,31,14,48,37,79,79,67,14,3,56,31,94,99,29,81,90,60,89,95,55,73,58,3,45,65,53,38,94,83,85,38,60,88,94,67,21,66,8,41,67,94,82,49,76,42,98,99,87,39,46,25,79,89,92,56,60,15,90,46,21,90,77,89,29,68,67,13,41,28,72,29,90,83,40,50,70,84,61,58,33,46,72,89,73,2,68,44,44,12,82,33,25,35,38,55,57,40,40,52,66,96,73,41,84,14,7,9,17,33,31,2,54,10,15,64,86,98,58,100,56,52,34,44,50,8,89,7,44,83,40,62,82,89,94,20,26,90,60,51,62,82,54,21,46,43,97,63,70,73,35,52,56,16,57,70,43,84,74,37,46,39,78,74,44,70,95,79,71,11,39,60,46,6,23,84,2,87,8,67,51,22,58,13,68,60,6,6,16,2,44,84,62,53,87,10,91,11,52,15,56,1,77,85,92,99,51,28,11,51,59,77,9,67,52,18,94,60,45,95,66,46,84,54,56,69,37,24,58,87,41,12,45,31,41,84,41,40,37,15,27,9,43,26,30,43,12,50,71,25,79,8,40,1,23,1,19,15,27,18,14,60,91,94,23,49,38,53,27,7,44,18,3,96,44,97,77,1,69,71,65,56,22,78,23,38,90,86,41,32,61,92,91,60,34,86,45,91,4,54,72,18,34,19,33,41,55,5,56,28,25,25,16,96,94,82,73,18,37,19,1,32,70,63,55,44,29,51,77,77,58,75,65,13,51,35,10,15,20,68,44,70,26,1,54,28,95,91,40,3,68,84,98,6,46,59,69,72,66,73,38,59,12,6,98,29,60,97,19,20,35,94,40,28,16,40,38,14,44,10,62,2,58,79,24,45,57,7,50,65,23,84,98,85,51,28,75,24,96,63,61,18,10,26,86,92,70,73,23,58,82,41,99,9,33,52,7,73,56,53,83,86,79,32,46,76,11,11,27,53,98,72,32,30,86,92,64,26,9,84,59,95,7,59,29,83,20,45,40,83,68,21,32,14,68,58,17,15,48,91,56,4,6,81,46,39,79,37,54,72,88,33,66,98,8,59,42,31,27,76,100,19,79,20,100,81,43,73,50,81,36,8,60,92,20,44,75,4,95,77,56,19,50,52,100,86,97,68,41,94,48,24,40,71,56,76,100,97,54,97,12,6,31,59,54,91,55,44,55,72,2,81,15,48,26,37,67,17,28,20,20,33,89,48,41,18,97,41,85,62,38,88,79,47,82,90,43,34,60,91,71,38,74,96,55,55,32,8,19,10,47,1,92,65,1,4,21,46,13,1,73,36,79,44,44,5,5,10,60,25,85,86,97,49,23,23,54,77,35,47,64,92,17,72,58,44,89,40,18,64,63,33,81,100,99,13,24,3,61,12,41,89,55,37,21,45,96,13,48,86,5,70,98,29,14,69,91,62,90,20,76,35,42,16,82,63,74,81,7,96,50,76,75,84,16,46,62,94,84,90,59,59,86,4,89,41,4,75,16,15,61,26,33,41,41,39,61,17,19,10,33,77,24,75,70,83,83,30,31,53,33,84,82,56,27,69,24,46,39,54,100,76,99,24,53,6,76,61,86,99,93,41,1,55,21,22,33,10,75,75,55,34,84,56,96,90,36,55,35,63,80,75,13,83,3,42,74,79,37,75,90,25,13,23,36,29,93,6,60,17,84,17,51,100,36,5,22,40,34,23,94,69,97,44,95,1,62,81,24,94,77,83,58,80,33,23,32,59,83,30,41,87,62,100,21,83,68 };
	signed short int y[1024] = { 25,48,40,60,81,11,83,85,36,44,58,71,75,76,39,43,96,58,85,28,63,59,97,9,51,53,10,91,89,44,79,15,62,27,45,85,20,31,49,34,80,99,16,24,71,38,98,98,65,87,41,64,99,56,94,73,49,64,89,20,40,100,41,66,91,100,66,11,4,62,57,97,75,67,53,26,97,55,4,70,52,6,90,34,23,12,32,23,66,7,28,29,89,45,76,61,79,12,98,85,6,47,33,64,24,58,61,60,45,4,52,41,11,46,46,56,81,71,88,6,22,46,96,80,46,34,6,75,51,20,43,17,76,37,95,2,83,63,54,66,73,10,88,2,30,18,93,7,59,64,66,87,6,82,53,70,22,55,71,96,45,9,6,63,80,70,35,95,53,96,8,21,78,92,79,30,16,85,79,28,23,33,83,83,58,58,29,70,80,45,45,47,28,68,91,91,75,27,69,14,13,20,15,59,8,83,73,93,50,66,90,54,29,98,4,33,98,37,31,13,92,14,34,90,50,62,59,70,3,53,4,83,34,85,25,59,94,5,6,3,69,60,12,80,62,8,7,14,79,10,24,25,11,86,70,74,66,52,33,67,12,15,2,97,98,13,47,66,30,76,56,43,27,76,90,73,41,94,26,54,96,27,26,93,7,30,60,21,64,80,51,66,80,24,61,12,52,84,93,50,28,66,92,51,98,20,12,30,40,43,32,70,10,41,30,31,11,60,29,16,1,29,56,88,5,91,14,84,81,92,14,51,41,18,58,61,22,52,99,49,70,42,4,30,81,35,9,52,37,74,53,81,82,19,13,83,64,2 [... cut ...] ,96,75,94,52,25,26,76,100,36,76,12,60,44,74,27,10,46,65,14,46,66,83,31 };

	// compute
	signed short int vecresult[1024];
	vec_add(vecresult, x, y, 1024);

	// stop
	*timer_data = 0x00000000;
	while (1);
}
```
The answer:

2ms delay when compiling with llvm
4ms delay when compiling with gcc
I know I may have done sth wrong attaching the timer (MAYBE) but the only difference was the change in the compiler when I was testing... So the compiler has this much impact on the final result!? I mean wow!

Mousavikia, 2022-06-04:
Hi @kuoyaoming93 ...  You said with runtime.h and utilizing get_mcycle(); I can count the number of cycles instead of utilizing my timer... I tried it with my code like this:
```
#include <uart.h>
#include <runtime.h>
#include <riscv_vector.h>
#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

void vec_add(signed short int* a, signed short int* b, signed short int* c, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e16m1(n);
		vint16m1_t vb = vle16_v_i16m1(b, vl);
		vint16m1_t vc = vle16_v_i16m1(c, vl);
		vint16m1_t va = vadd_vv_i16m1(vb, vc, vl);
		vse16_v_i16m1(a, va, vl);
		a += vl;
		b += vl;
		c += vl;
		n -= vl;
	}

}
int main(void) {
	//start
	long time1,time2;
	time1 = get_mcycle();
	// data
	signed short int x[1024] = { 19,5,73,35,67,39,63,3,92,81,75,82,39,62,58,54,28,25,46,23,81,99,3,54,9,81,99,7,94,2,69,79,54,89,90,63,14,22,19,5,11,62,94,36,42,99,95,68,99,77,34,67,25,30,69,53,42,61,76,59,56,59,52,9,72,100,36,98,35,89,46,42,22,13,31,73,79,70,1,85,93,78,5,38,71,73,23,27,68,48,63,24,18,83,77,94,11,19,10,49,20,90,10,5,56,78,32,18,34,22,52,91,63,11,40,6,51,44,100,82,49,90,14,40,93,92,72,62,35,94,13,74,65,84,40,75,84,33,56,98,55,34,62,37,76,42,50,70,98,33,84,74,96,4,36,67,29,24,72,63,60,67,5,35,46,25,72,86,29,74,14,84,14,59,37,81,51,49,88,36,45,97,5,98,19,67,59,68,37,63,82,2,9,98,66,24,41,13,27,26,34,16,35,13,89,10,94,40,5,35,74,80,55,69,90,6,31,5,20,73,73,88,59,8,93,81,29,55,99,72,84,44,48,57,27,75,51,65,31,14,48,37,79,79,67,14,3,56,31,94,99,29,81,90,60,89,95,55,73,58,3,45,65,53,38,94,83,85,38,60,88,94,67,21,66,8,41,67,94,82,49,76,42,98,99,87,39,46,25,79,89,92,56,60,15,90,46,21,90,77,89,29,68,67,13,41,28,72,29,90,83,40,50,70,84,61,58,33,46,72,89,73,2,68,44,44,12,82,33,25,35,38,55,57,40,40,52,66,96,73,41,84,14,7,9,17,33,31,2,54,10,15,64,86,98,58,100,56,52,34,44,50,8,89,7,44,83,40,62,82,89,94,20,26,90,60,51,62,82,54,21,46,43,97,63,70,73,35,52,56,16,57,70,43,84,74,37,46,39,78,74,44,70,95,79,71,11,39,60,46,6,23,84,2,87,8,67,51,22,58,13,68,60,6,6,16,2,44,84,62,53,87,10,91,11,52,15,56,1,77,85,92,99,51,28,11,51,59,77,9,67,52,18,94,60,45,95,66,46,84,54,56,69,37,24,58,87,41,12,45,31,41,84,41,40,37,15,27,9,43,26,30,43,12,50,71,25,79,8,40,1,23,1,19,15,27,18,14,60,91,94,23,49,38,53,27,7,44,18,3,96,44,97,77,1,69,71,65,56,22,78,23,38,90,86,41,32,61,92,91,60,34,86,45,91,4,54,72,18,34,19,33,41,55,5,56,28,25,25,16,96,94,82,73,18,37,19,1,32,70,63,55,44,29,51,77,77,58,75,65,13,51,35,10,15,20,68,44,70,26,1,54,28,95,91,40,3,68,84,98,6,46,59,69,72,66,73,38,59,12,6,98,29,60,97,19,20,35,94,40,28,16,40,38,14,44,10,62,2,58,79,24,45,57,7,50,65,23,84,98,85,51,28,75,24,96,63,61,18,10,26,86,92,70,73,23,58,82,41,99,9,33,52,7,73,56,53,83,86,79,32,46,76,11,11,27,53,98,72,32,30,86,92,64,26,9,84,59,95,7,59,29,83,20,45,40,83,68,21,32,14,68,58,17,15,48,91,56,4,6,81,46,39,79,37,54,72,88,33,66,98,8,59,42,31,27,76,100,19,79,20,100,81,43,73,50,81,36,8,60,92,20,44,75,4,95,77,56,19,50,52,100,86,97,68,41,94,48,24,40,71,56,76,100,97,54,97,12,6,31,59,54,91,55,44,55,72,2,81,15,48,26,37,67,17,28,20,20,33,89,48,41,18,97,41,85,62,38,88,79,47,82,90,43,34,60,91,71,38,74,96,55,55,32,8,19,10,47,1,92,65,1,4,21,46,13,1,73,36,79,44,44,5,5,10,60,25,85,86,97,49,23,23,54,77,35,47,64,92,17,72,58,44,89,40,18,64,63,33,81,100,99,13,24,3,61,12,41,89,55,37,21,45,96,13,48,86,5,70,98,29,14,69,91,62,90,20,76,35,42,16,82,63,74,81,7,96,50,76,75,84,16,46,62,94,84,90,59,59,86,4,89,41,4,75,16,15,61,26,33,41,41,39,61,17,19,10,33,77,24,75,70,83,83,30,31,53,33,84,82,56,27,69,24,46,39,54,100,76,99,24,53,6,76,61,86,99,93,41,1,55,21,22,33,10,75,75,55,34,84,56,96,90,36,55,35,63,80,75,13,83,3,42,74,79,37,75,90,25,13,23,36,29,93,6,60,17,84,17,51,100,36,5,22,40,34,23,94,69,97,44,95,1,62,81,24,94,77,83,58,80,33,23,32,59,83,30,41,87,62,100,21,83,68 };
	signed short int y[1024] = { 25,48,40,60,81,11,83,85,36,44,58,71,75,76,39,43,96,58,85,28,63,59,97,9,51,53,10,91,89,44,79,15,62,27,45,85,20,31,49,34,80,99,16,24,71,38,98,98,65,87,41,64,99,56,94,73,49,64,89,20,40,100,41,66,91,100,66,11,4,62,57,97,75,67,53,26,97,55,4,70,52,6,90,34,23,12,32,23,66,7,28,29,89,45,76,61,79,12,98,85,6,47,33,64,24,58,61,60,45,4,52,41,11,46,46,56,81,71,88,6,22,46,96,80,46,34,6,75,51,20,43,17,76,37,95,2,83,63,54,66,73,10,88,2,30,18,93,7,59,64,66,87,6,82,53,70,22,55,71,96,45,9,6,63,80,70,35,95,53,96,8,21,78,92,79,30,16,85,79,28,23,33,83,83,58,58,29,70,80,45,45,47,28,68,91,91,75,27,69,14,13,20,15,59,8,83,73,93,50,66,90,54,29,98,4,33,98,37,31,13,92,14,34,90,50,62,59,70,3,53,4,83,34,85,25,59,94,5,6,3,69,60,12,80,62,8,7,14,79,10,24,25,11,86,70,74,66,52,33,67,12,15,2,97,98,13,47,66,30,76,56,43,27,76,90,73,41,94,26,54,96,27,26,93,7,30,60,21,64,80,51,66,80,24,61,12,52,84,93,50,28,66,92,51,98,20,12,30,40,43,32,70,10,41,30,31,11,60,29,16,1,29,56,88,5,91,14,84,81,92,14,51,41,18,58,61,22,52,99,49,70,42,4,30,81,35,9,52,37,74,53,81,82,19,13,83,64,2,90,52,55,61,77,86,39,9,74,34,84,38,83,18,13,88,5,69,74,44,38,98,40,45,16,33,32,90,25,32,41,71,15,88,9,47,4,76,71,22,68,56,86,56,91,42,36,49,26,93,47,26,44,71,41,19,86,59,38,23,22,53,44,75,8,85,68,14,86,20,61,55,17,1,78,77,43,6,59,18,73,54,26,92,76,89,7,19,74,70,78,51,43,62,86,68,53,30,71,39,57,89,85,90,94,82,1,1,9,27,3,43,35,55,93,30,34,86,35,14,51,86,39,70,63,46,48,95,9,28,45,59,88,47,44,75,47,87,47,50,49,23,9,7,89,24,87,72,88,94,14,40,99,65,90,49,2,63,24,53,73,61,59,44,25,43,2,61,96,10,4,89,25,1,82,15,88,10,36,60,59,67,65,44,14,76,25,66,86,9,98,4,84,84,5,55,95, [... cut ...] ,27,60,48,37,66,94,63,29,21,44,3,88,62,21,52,6,87,45,55,57,69,38,8,46,5,74,4,96,75,94,52,25,26,76,100,36,76,12,60,44,74,27,10,46,65,14,46,66,83,31 };

	// compute
	signed short int vecresult[1024];
	vec_add(vecresult, x, y, 1024);

	// stop
	time2 = get_mcycle();
	uart_printf("The operation took %d cycles\n\r",time2-time1);
	while (1);
}
```
But it gives an error when I compile the code:
[image]

Note: I have updated the repo to the newest version so I have the header files inside the lib folder

kuoyaoming93, 2022-06-04:
HI @Mousavikia,

The runtime function is noot included in compilation. See the line 38 in **sw/Makefile**.

https://github.com/vproc/vicuna/blob/8511bc0fd930a5100bdd82cb46a4e9a90be240c2/sw/Makefile#L38

Just add runtime.o like this:

`
RISCV_OBJ := $(SW_DIR)/crt0.o $(OBJ) $(SW_DIR)/lib/uart.o $(SW_DIR)/lib/runtime.o
`

My suggestion:
I would measure the time when the vector extensions are used, the initialization of the vectors is part of C code, and not an acceleration using the vector extension.
I don't know if the data initialization should be part of it. 

Example:
```c
int main(void) {
	// data
	signed short int x[1024] = { 19,5,73,35,67,39,63,3,92,81,75,82,39,62,...};
	signed short int y[1024] = { 25,48,40,60,81,11,83,85,36,44,58,71,75,76,39,43,96,58,85,28,63,59,97... };

        //start
	long time1,time2;
	time1 = get_mcycle();

	// compute
	signed short int vecresult[1024];
	vec_add(vecresult, x, y, 1024);

	// stop
	time2 = get_mcycle();
	uart_printf("The operation took %d cycles\n\r",time2-time1);
	while (1);
}
```

Mousavikia, 2022-06-04:
Thank you @kuoyaoming93 the error is gone...
About the initialization, I understand what you are talking. Let's say for a fair comparison between normal code (no vector) and vector mode code we should initialize the timer (or cycle count here) at the same time (for example after or before data initialization)... In this case, probably activating the counter after initialization for both cases (vector and non-vector) can be better as you suggest

kuoyaoming93, 2022-06-08:
Hi @Mousavikia 

I can do the verilator simulation with get_mcycle() function using the cv32e40x core. Just change the line 26 in simulation Makefile

`CORE_DIR ?= $(SIM_DIR)/../cv32e40x/`

https://github.com/vproc/vicuna/blob/60ec4a7333734e8ee0669431194217c155306186/sim/Makefile#L26

michael-platzer, 2022-06-08:
According to [the documentation of Ibex](https://ibex-core.readthedocs.io/en/latest/03_reference/performance_counters.html) the performance counters should always be enabled. I have no idea why this appears not to be working in Verilator.

Mousavikia, 2022-06-10:
Hi @michael-platzer and @kuoyaoming93 I don't know what have you changed but now in verilator and using Ibex I can use get_mcycle() but there is sth wrong. No matter what configuration I use the cycle count is same:
My code:
[image]
Result: (compact configuration, triple and my added configuration (VMEM_W=128 and VREG_W=256)):
[image]

closed 2022-06-27 by Mousavikia