vproc/vicuna #72: Question about vicuna simulation directory

Mousavikia, 2022-06-04:
Hi sorry I am asking lots of questions today but in simulation directory you stated this:
`The files in this directory allow to simulate the vector processor. Simulation is done with the help of make. The environment variable PROG_PATHS_LIST can be used to specify a series of programs that should be executed on the simulated vector processor. Such a list of programs can be a simple text file with each line containing the file path of a *.vmem file specifying the memory initialization file of the respective programs.`
I tried to simulate the vector add code like this:
I created a file name `progs.csv` in the sim directory and inside that I introduced the path to .vmem file like this:
/home/kiian/Desktop/VicunaRepos/632022/vicuna/sim/TestPrograms/vecaddtimer.vmem
Then I ran `make PROG_PATHS_LIST=progs.csv` in the sim directory.
The output is a `sim_trace.csv` file and inside it is:
`rst_ni;mem_req;mem_addr;pend_vreg_wr_map_o;`
Which is nothing... what am I doing wrong?

Mousavikia, 2022-06-05:
@kuoyaoming93 can you help? I cannot find the results of the simulation ofcourse if I am doing it right!

kuoyaoming93, 2022-06-05:
Hi @Mousavikia,

By output do you mean VCD timing diagram? 

Because you can generate the VCD file using the following command:
`
make verilator TRACE_VCD=<test_vcd_path>.vcd
`

If you want to see the console output, you can  just use uart_printf provided in the libraries. For example:

```c
#include <uart.h>

int main(void)
{
    uart_printf("Hello World from Vicuna! \n\r");

    asm volatile("jr x0;");
    return 0;
}
```

When you run `make verilator`, you will see this in the console:

[image]

Mousavikia, 2022-06-05:
Ok let me explain so that you know how basic I am in this things :(( Sorry!
I created a test.c program in sw directory containing your code:
```
#include <uart.h>

int main(void)
{
    uart_printf("Hello World from Vicuna! \n\r");

    asm volatile("jr x0;");
    return 0;
}
```
Then I compiled the code using gcc and now I have the .vmem file in sw directory.
Then in sim directory I created a progs.csv file and inside that I placed the path to my .vmem file like this:
`/home/kiian/Desktop/VicunaRepos/632022/vicuna_verilator/sw/test.vmem`
Finally in sim directory in terminal I typed `make verilator ` and I see this output in terminal:
[image]
You have the same (in Spanish I suppose) but you have the output but I don't...

kuoyaoming93, 2022-06-05:
It's ok @Mousavikia, no problem, we are all learning together :)

Are you using the last repository commit 2331baac7ff6f2ef0e72294993198a0d356cd3f2? 
Because some changes were made by @michael-platzer to make the verilator cpp compatible with uart_printf.

Please do `git pull` to the repository and try it again.

Mousavikia, 2022-06-05:
Thank you @kuoyaoming93 yeah I did update the repository right now before sending the last comment but steal nothing....

kuoyaoming93, 2022-06-05:
@Mousavikia 

In prog.csv put this:
`<PATH_TO_VMEM>/test.vmem /dev/null 0 0 /dev/null 0 0`

The reason of the arguments are explained here: https://github.com/vproc/vicuna/issues/66#issuecomment-1141061928

Mousavikia, 2022-06-05:
You are a hero my brother...
[image]

Mousavikia, 2022-06-05:
Before I bother you again tomorrow :)) just one more thing:
with simulation, if I give this code:
```
#include <uart.h>

#include <runtime.h>

#include <riscv_vector.h>

#include <stddef.h>

void* memcpy(void* dest, const void* src, size_t n)

{

	for (size_t i = 0; i < n; i++)

	{

		((char*)dest)[i] = ((char*)src)[i];

	}

}

void vec_add(signed short int* a, signed short int* b, signed short int* c, signed short int n) {

	while (n > 0) {

		size_t vl = vsetvl_e16m1(n);

		vint16m1_t vb = vle16_v_i16m1(b, vl);

		vint16m1_t vc = vle16_v_i16m1(c, vl);

		vint16m1_t va = vadd_vv_i16m1(vb, vc, vl);

		vse16_v_i16m1(a, va, vl);

		a += vl;

		b += vl;

		c += vl;

		n -= vl;

	}

}

int main(void) {

	//start

	long time1,time2;

	time1 = get_mcycle();

	// data

	signed short int x[1024] = { 19,5,73,35,67,39,63,3,92,81,75,82,39,62,58,54,28,25,46,23,81,99,3,54,9,81,99,7,94,2,69,79,54,89,90,63,14,22,19,5,11,62,94,36,42,99,95,68,99,77,34,67,25,30,69,53,42,61,76,59,56,59,52,9,72,100,36,98,35,89,46,42,22,13,31,73,79,70,1,85,93,78,5,38,71,73,23,27,68,48,63,24,18,83,77,94,11,19,10,49,20,90,10,5,56,78,32,18,34,22,52,91,63,11,40,6,51,44,100,82,49,90,14,40,93,92,72,62,35,94,13,74,65,84,40,75,84,33,56,98,55,34,62,37,76,42,50,70,98,33,84,74,96,4,36,67,29,24,72,63,60,67,5,35,46,25,72,86,29,74,14,84,14,59,37,81,51,49,88,36,45,97,5,98,19,67,59,68,37,63,82,2,9,98,66,24,41,13,27,26,34,16,35,13,89,10,94,40,5,35,74,80,55,69,90,6,31,5,20,73,73,88,59,8,93,81,29,55,99,72,84,44,48,57,27,75,51,65,31,14,48,37,79,79,67,14,3,56,31,94,99,29,81,90,60,89,95,55,73,58,3,45,65,53,38,94,83,85,38,60,88,94,67,21,66,8,41,67,94,82,49,76,42,98,99,87,39,46,25,79,89,92,56,60,15,90,46,21,90,77,89,29,68,67,13,41,28,72,29,90,83,40,50,70,84,61,58,33,46,72,89,73,2,68,44,44,12,82,33,25,35,38,55,57,40,40,52,66,96,73,41,84,14,7,9,17,33,31,2,54,10,15,64,86,98,58,100,56,52,34,44,50,8,89,7,44,83,40,62,82,89,94,20,26,90,60,51,62,82,54,21,46,43,97,63,70,73,35,52,56,16,57,70,43,84,74,37,46,39,78,74,44,70,95,79,71,11,39,60,46,6,23,84,2,87,8,67,51,22,58,13,68,60,6,6,16,2,44,84,62,53,87,10,91,11,52,15,56,1,77,85,92,99,51,28,11,51,59,77,9,67,52,18,94,60,45,95,66,46,84,54,56,69,37,24,58,87,41,12,45,31,41,84,41,40,37,15,27,9,43,26,30,43,12,50,71,25,79,8,40,1,23,1,19,15,27,18,14,60,91,94,23,49,38,53,27,7,44,18,3,96,44,97,77,1,69,71,65,56,22,78,23,38,90,86,41,32,61,92,91,60,34,86,45,91,4,54,72,18,34,19,33,41,55,5,56,28,25,25,16,96,94,82,73,18,37,19,1,32,70,63,55,44,29,51,77,77,58,75,65,13,51,35,10,15,20,68,44,70,26,1,54,28,95,91,40,3,68,84,98,6,46,59,69,72,66,73,38,59,12,6,98,29,60,97,19,20,35,94,40,28,16,40,38,14,44,10,62,2,58,79,24,45,57,7,50,65,23,84,98,85,51,28,75,24,96,63,61,18,10,26,86,92,70,73,23,58,82,41,99,9,33,52,7,73,56,53,83,86,79,32,46,76,11,11,27,53,98,72,32,30,86,92,64,26,9,84,59,95,7,59,29,83,20,45,40,83,68,21,32,14,68,58,17,15,48,91,56,4,6,81,46,39,79,37,54,72,88,33,66,98,8,59,42,31,27,76,100,19,79,20,100,81,43,73,50,81,36,8,60,92,20,44,75,4,95,77,56,19,50,52,100,86,97,68,41,94,48,24,40,71,56,76,100,97,54,97,12,6,31,59,54,91,55,44,55,72,2,81,15,48,26,37,67,17,28,20,20,33,89,48,41,18,97,41,85,62,38,88,79,47,82,90,43,34,60,91,71,38,74,96,55,55,32,8,19,10,47,1,92,65,1,4,21,46,13,1,73,36,79,44,44,5,5,10,60,25,85,86,97,49,23,23,54,77,35,47,64,92,17,72,58,44,89,40,18,64,63,33,81,100,99,13,24,3,61,12,41,89,55,37,21,45,96,13,48,86,5,70,98,29,14,69,91,62,90,20,76,35,42,16,82,63,74,81,7,96,50,76,75,84,16,46,62,94,84,90,59,59,86,4,89,41,4,75,16,15,61,26,33,41,41,39,61,17,19,10,33,77,24,75,70,83,83,30,31,53,33,84,82,56,27,69,24,46,39,54,100,76,99,24,53,6,76,61,86,99,93,41,1,55,21,22,33,10,75,75,55,34,84,56,96,90,36,55,35,63,80,75,13,83,3,42,74,79,37,75,90,25,13,23,36,29,93,6,60,17,84,17,51,100,36,5,22,40,34,23,94,69,97,44,95,1,62,81,24,94,77,83,58,80,33,23,32,59,83,30,41,87,62,100,21,83,68 };

	signed short int y[1024] = { 25,48,40,60,81,11,83,85,36,44,58,71,75,76,39,43,96,58,85,28,63,59,97,9,51,53,10,91,89,44,79,15,62,27,45,85,20,31,49,34,80,99,16,24,71,38,98,98,65,87,41,64,99,56,94,73,49,64,89,20,40,100,41,66,91,100,66,11,4,62,57,97,75,67,53,26,97,55,4,70,52,6,90,34,23,12,32,23,66,7,28,29,89,45,76,61,79,12,98,85,6,47,33,64,24,58,61,60,45,4,52,41,11,46,46,56,81,71,88,6,22,46,96,80,46,34,6,75,51,20,43,17,76,37,95,2,83,63,54,66,73,10,88,2,30,18,93,7,59,64,66,87,6,82,53,70,22,55,71,96,45,9,6,63,80,70,35,95,53,96,8,21,78,92,79,30,16,85,79,28,23,33,83,83,58,58,29,70,80,45,45,47,28,68,91,91,75,27,69,14,13,20,15,59,8,83,73,93,50,66,90,54,29,98,4,33,98,37,31,13,92,14,34,90,50,62,59,70,3,53,4,83,34,85,25,59,94,5,6,3,69,60,12,80,62,8,7,14,79,10,24,25,11,86,70,74,66,52,33,67,12,15,2,97,98,13,47,66,30,76,56,43,27,76,90,73,41,94,26,54,96,27,26,93,7,30,60,21,64,80,51,66,80,24,61,12,52,84,93,50,28,66,92,51,98,20,12,30,40,43,32,70,10,41,30,31,11,60,29,16,1,29,56,88,5,91,14,84,81,92,14,51,41,18,58,61,22,52,99,49,70,42,4,30,81,35,9,52,37,74,53,81,82,19,13,83,64,2,90,52,55,61,77,86,39,9,74,34,84,38,83,18,13,88,5,69,74,44,38,98,40,45,16,33,32,90,25,32,41,71,15,88,9,47,4,76,71,22,68,56,86,56,91,42,36,49,26,93,47,26,44,71,41,19,86,59,38,23,22,53,44,75,8,85,68,14,86,20,61,55,17,1,78,77,43,6,59,18,73,54,26,92,76,89,7,19,74,70,78,51,43,62,86,68,53,30,71,39,57,89,85,90,94,82,1,1,9,27,3,43,35,55,93,30,34,86,35,14,51,86,39,70,63,46,48,95,9,28,45,59,88,47,44,75,47,87,47,50,49,23,9,7,89,24,87,72,88,94,14,40,99,65,90,49,2,63,24,53,73,61,59,44,25,43,2,61,96,10,4,89,25,1,82,15,88,10,36,60,59,67,65,44,14,76,25,66,86,9,98,4,84,84,5,55,95,33,81,61,79,80,5,29,66,49,98,75,57,30,26,89,45,82, [... cut ...] 0,47,57,50,7,90,29,27,60,48,37,66,94,63,29,21,44,3,88,62,21,52,6,87,45,55,57,69,38,8,46,5,74,4,96,75,94,52,25,26,76,100,36,76,12,60,44,74,27,10,46,65,14,46,66,83,31 };

	// compute

	signed short int vecresult[1024];

	vec_add(vecresult, x, y, 1024);

	// stop

	time2 = get_mcycle();

	uart_printf("The operation took %d cycles\n\r",time2-time1);

	while (1);

}
```
Both simulation and FPGA will give the same result?
How does verilator know about clk frequency?
Or it will give approximate values?

kuoyaoming93, 2022-06-05:
I have to find out why get_mcycle() does not work with Verilator simulation. See discussion here: https://github.com/vproc/vicuna/issues/66#issuecomment-1143403522
The function is supported in **FPGA only** now. I think there is configuration mismatch between verilator and FPGA.

Using that function, you only get the number of cycles, it should be similar to implementation if the configurations and memory hierarchy are the same. 

The best accurate way is to implement it to FPGA and determinate the clock cycles required for the whole system.

Mousavikia, 2022-06-05:
Yeah, I was trying to do the same thing and it doesn't work for me as well...
[image]
That would be really nice if we have that feature in verilator since implementation on FPGA is very very time-consuming, especially with outdated computers (like mine)...

Mousavikia, 2022-06-06:
Hi again @kuoyaoming93 and @michael-platzer... Utilizing verilator I simulated this code:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>
void* memcpy(void* dest, const void* src, size_t n)
{

	for (size_t i = 0; i < n; i++)

	{
		((char*)dest)[i] = ((char*)src)[i];

	}
}
int compare(signed short int* ref, signed short int* actual, signed short int n) {
	signed short int r;
	for (signed short int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}
void vec_mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vint16m2_t vout = vwmul_vv_i16m2(vin1, vin2, vl);
		vse16_v_i16m2(out, vout, vl);
		in1 += vl;
		in2 += vl;
		out += vl;
		n -= vl;
	}
}
void mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	signed short int i;
	for (i = 0; i < n; i++) {
		out[i] = in1[i] * in2[i];
	}
}
int main(void) {
	// data
	signed char a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	signed char b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	// compute
	signed short int result[31];
	signed short int vecresult[31];
	mul(result, a, b, 31);
	vec_mul(vecresult, a, b, 31);
	// compare
	uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
   	asm volatile("jr x0;");
	return 0;
}
```
But there is a problem...
Verilator returns fail in the console but with FPGA implementation uart returns pass!!
Do you know what can be the problem?

commit 883aa037 (2022-06-06): Fix parsing of program paths file in simulation

Allow a program paths file with lines containing just the memory
initialization file (no dump files).  Change its default name to
`progs.txt` and rename the associated environment variable to
`PROG_PATHS`.  Fixes #72.