vproc/vicuna #74: Vector multipilication output is incorrect

Mousavikia, 2022-06-07:
Hi @michael-platzer , @kuoyaoming93. Sorry that I opened another issue since the question I ask now is the same that I asked in [https://github.com/vproc/vicuna/issues/72](url)... My main concern is whether my code is wrong or there is a bug in hardware since in [https://github.com/vproc/vicuna/issues/55](url) @stevobailey mentioned sth about unaligned memory address and said that is probably why vredsum in my case is not working... Anyway, I carefully wrote a multiplication code both in vector intrinsic mode and normal mode:
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>
void* memcpy(void* dest, const void* src, size_t n)
{

	for (size_t i = 0; i < n; i++)

	{
		((char*)dest)[i] = ((char*)src)[i];

	}
}
int compare(signed short int* ref, signed short int* actual, signed short int n) {
	signed short int r;
	for (signed short int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}
void vec_mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vint16m2_t vout = vwmul_vv_i16m2(vin1, vin2, vl);
		vse16_v_i16m2(out, vout, vl);
		in1 += vl;
		in2 += vl;
		out += vl;
		n -= vl;
	}
}
void mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	signed short int i;
	for (i = 0; i < n; i++) {
		out[i] = in1[i] * in2[i];
	}
}
int main(void) {
	// data
	signed char a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	signed char b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
	// compute
	signed short int result[31];
	signed short int vecresult[31];
	mul(result, a, b, 31);
	vec_mul(vecresult, a, b, 31);
	// compare
 	for(int i = 0; i<31; i++){
 		uart_printf("The result of vector mul is: %d \n",vecresult[i]);
	}:
   	asm volatile("jr x0;");
	return 0;
}
```
Then I both simulated and implemented the code on my FPGA. For normal multiplication everything is right but the vector results are incorrect. You can see them in:
[image]
Sorry that I am insisting but since I am trying to finish the final part of my thesis "comparing vector and normal processor" I am a bit worried about the incorrect outputs.

stevobailey, 2022-06-07:
> Hi @michael-platzer , @kuoyaoming93. Sorry that I opened another issue since the question I ask now is the same that I asked in https://github.com/vproc/vicuna/issues/72... My main concern is whether my code is wrong or there is a bug in hardware since in https://github.com/vproc/vicuna/issues/55 @stevobailey mentioned sth about unaligned memory address and said that is probably why vredsum in my case is not working... Anyway, I carefully wrote a multiplication code both in vector intrinsic mode and normal mode:

A quick way to check if alignment is the issue is to rewrite your code to use 32-bit data instead of 8-bit data. If it is correct with 32-bit data, then it’s probably the alignment issue.

michael-platzer, 2022-06-07:
Hi @Mousavikia,

you can check the alignment of your arrays by inspecting the `*.elf` file. Type following command to see the addresses of symbols:
```
readelf -s prog.elf
```

You will have to move the declaration of the arrays out of your function (making them global variables) to be able to see them in the output. Now search for `a` and `b`, and for `vecresult` as well (symbol names are in the last column and the address is in the 2nd column, called `Value`). If these are not aligned to the width of the memory interface, then this is almost certainly the source of the error.

Please use the `align` attribute to align your arrays. The alignment (specified in bytes) must be at least the byte width of the memory interface:
```
signed char a[31] __attribute__ ((aligned (16))) = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
```

Mousavikia, 2022-06-07:
It is really blessing to have you guys answering all of the questions...
Thank you @stevobailey and @michael-platzer ....
I first did this (make the arrays global variables):
```
#include <uart.h>
#include <riscv_vector.h>
#include <stddef.h>
// data
signed char a[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
signed char b[31] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };
// compute
signed short int result[31];
signed short int vecresult[31];
void* memcpy(void* dest, const void* src, size_t n)
{
	for (size_t i = 0; i < n; i++)
	{
		((char*)dest)[i] = ((char*)src)[i];
	}
}

int compare(signed short int* ref, signed short int* actual, signed short int n) {
	signed short int r;
	for (signed short int i = 0; i < n; ++i) {
		if (ref[i] - actual[i] == 0) {
			r = 1;
		}
		else {
			r = 0;
			break;
		}
	}
	return r;
}

void vec_mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	while (n > 0) {
		size_t vl = vsetvl_e8m1(n);
		vint8m1_t vin1 = vle8_v_i8m1(in1, vl);
		vint8m1_t vin2 = vle8_v_i8m1(in2, vl);
		vint16m2_t vout = vwmul_vv_i16m2(vin1, vin2, vl);
		vse16_v_i16m2(out, vout, vl);
		in1 += vl;
		in2 += vl;
		out += vl;
		n -= vl;
	}
}

void mul(signed short int* out, signed char* in1, signed char* in2, signed short int n) {
	signed short int i;
	for (i = 0; i < n; i++) {
		out[i] = in1[i] * in2[i];
	}
}

int main(void) {
	mul(result, a, b, 31);
	vec_mul(vecresult, a, b, 31);
	// compare
	for(int i = 0; i<31; i++){
 		uart_printf("The result of vector mul is: %d \n",vecresult[i]);
	}
	uart_printf("\n\n");
	for(int i = 0; i<31; i++){
 		uart_printf("The result of normal mul is: %d \n",result[i]);
	}
	//uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
   	asm volatile("jr x0;");
	return 0;
[... 261 lines cut ...]
	// compute
	signed short int result[31];
	signed short int vecresult[31];
	mul(result, a, b, 31);
	vec_mul(vecresult, a, b, 31);
	// compare
	for(int i = 0; i<31; i++){
 		uart_printf("The result of vector mul is: %d \n",vecresult[i]);
	}
	uart_printf("\n\n");
	for(int i = 0; i<31; i++){
 		uart_printf("The result of normal mul is: %d \n",result[i]);
	}
	//uart_puts(compare(vecresult, result, 31) ? "pass" : "fail");
   	asm volatile("jr x0;");
	return 0;
}
```
And the outputs:
[image]
[image]
They are match :)
Just one thing... I don't really know much about this memory unaligned thing but you provided this (`signed char a[31] __attribute__ ((aligned (16))) = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31 };`) for variable a and I used it for b and not for vecresult and result variable and it worked. My question is this:
The memory bus is 32 but you passed 16 in there? If the variables is signed short int what should I do? Should I  pass 32 to aligned)? And I shouldn't use all this if I want to have int data type(since int match memory width)?! Right? Can you explain a bit more about this concept?
Thank you really much for the time you are expending....

michael-platzer, 2022-06-07:
The alignment for the aligned attribute is specified in bytes, whereas the width of Vicuna's memory interface (`VMEM_W`) is in bits. An alignment of 16 bytes (=128 bits) ensures that the base address is aligned for memory interface widths up to that value (i.e., a `VMEM_W` up to 128 bits). For a 32-bit memory interface an alignment of 4 bytes (`__attribute__ ((aligned (4))`) is enough.

The alignment issue is a limitation that is actually not spec compliant, and given recent changes in the XIF interface specification it will be the coprocessor's task to align memory data. Therefore, I am planning to implement some alignment logic that will lift this limitation soon.

Mousavikia, 2022-06-08:
Great... Thank you so much @michael-platzer.

michael-platzer, 2022-06-09:
Hi @Mousavikia, @stevobailey,

to resolve the problem of unit-strided loads/stores that are not aligned to the width of the memory interface, Vicuna's decoder does now convert unit-strided loads/stores with an unsuitably aligned base address to generic strided loads/stores (with a stride of 1) in daa79d4. That way, the behavior is correct. However, the strided loads/stores are much less efficient than the unit-strided ones, which are only used for suitably aligned base addresses.

To override this behavior (and use the more efficient unit-strided variant for all unit-strided loads/stores, with potentially incorrect results), the flag `VLSU_ALIGNED_UNITSTRIDE` can be removed from the `VLSU_FLAGS` in `vproc_top.sv`:

https://github.com/vproc/vicuna/blob/1b32c1304969e559f07dd03070711950f140ce2e/rtl/vproc_top.sv#L401-L402

closed 2022-06-09 by michael-platzer