vproc/vicuna #75: vzext.vf2 instruction execution problem

moimfeld, 2022-06-07:
Hi @michael-platzer 

When executing the vzext.vf2 instruction on Vicuna the UVM environment issues the following warning (and error):

    # UVM_WARNING uvm/src/env/cvxif_scoreboard.svh(207) @ 2430: uvm_test_top.env.scoreboard [CVXIF_SCOREBOARD] Instr "vzext.vf2 v4, v0 e16, m1, tu, mu" in Prog "vzext_8" failed: 
    # at                 2430: State Difference
    # ref.vproc_register[4]: 00c70043004b003a0032003b00bf0047, dut.vproc_reg[4]: 0000c74300004b3a0000323b0000bf47
    # 
    # 
    # UVM_ERROR uvm/src/env/cvxif_scoreboard.svh(223) @ 2580: uvm_test_top.env.scoreboard [CVXIF_SCOREBOARD] Program "vzext_8" failed

It looks like Vicuna zero-extends the elements from SEW to 2*SEW while spike zero-extends the elements from SEW/2 to SEW. If I understand the vector specs correctly, then Spike executes the instruction correctly.
You can reproduce this error by running the `cvxif_test_direct_issue_75` test in my UVM environment.

Sidenote: these are the vector instructions executed during the program vzext_8 (note that SEW = 16):

    vsetvli         t0, t0, e16,m1,tu,mu
    vle8.v          v0, (a0)
    vzext.vf2       v4, v0
    vse16.v         v4, (a0)

stevobailey, 2022-07-25:
Similarly, I have clang auto-vectorizing code, and it creates code that uses the `vzext.vf4` instruction. But running this on Vicuna results in an illegal instruction. I only see support for `vzext.vf2` and `vsext.vf2`. Should there be support for `v[s|z]ext.vf[4|8]`? Or is it an issue with auto-vectorization or my compiler settings?

moimfeld, 2022-07-25:
Hi @stevobailey 
AFAIK, to be compliant with `Zve32x` there should be support for `*.vf[2|4]`, where `vf4` is only legal if `SEW` = 32 (because the instruction will extend from elements `SEW/4` to `SEW`). As you pointed out, there is only support for `v[s|z]ext.vf2` at the moment.

If the `vzext.vf4` instruction created by your compiler is executed while `SEW` = 32 then there should be nothing wrong with your compiler.

stevobailey, 2022-07-25:
Got it, thanks! Then it is a related issue:
https://github.com/vproc/vicuna/blob/bda35112a983e9adb4ea8413b0afc2f0d91e7563/rtl/vproc_decoder.sv#L648

mahnaz-namazi, 2023-06-12:
Hi @moimfeld @michael-platzer 

has this bug been solved or should we use other instructions than vzext.vf2 for now?
this instruction is executed correctly by spike but there is "illegal instruction" error when using vicuna for RTL simulation.

thanks