vproc/vicuna #76: [Question] Data cache configuration in demo_top

kuoyaoming93, 2022-06-08:
Hi all, I have a question regarding data cache in demo_top example. 

1. Is there any specific configuration I have to do to use data cache in demo_top?
2. Is there any advantage to use data cache in this example? The memories are BRAMs and not external memories.

michael-platzer, 2022-06-08:
Hi @kuoyaoming93,

> Is there any specific configuration I have to do to use data cache in demo_top?

The data cache can be enabled by setting the `vproc_top` module's parameter `DCACHE_SZ` to a different value than 0. Note that this parameter specifies the size in bytes of the data cache, whereas the line width `DCACHE_LINE_W` as well as the width of the memory interfaces are given in bits.

> Is there any advantage to use data cache in this example? The memories are BRAMs and not external memories.

No, the data cache only makes sense if you have an external memory with long access latency. Onboard BRAM that has 1 cycle access latency is as fast as the data cache.

kuoyaoming93, 2022-06-08:
Thank you @michael-platzer!! All clear

closed 2022-06-08 by kuoyaoming93