vproc/vicuna #81: vrgather.vv issue (possible control signal issue in vproc_elem)

moimfeld, 2022-06-12:
Hello @michael-platzer 

## Issue

When enabling random delay for the memory result transactions (w.r.t. to the memory issue transaction) an issue with the vrgather.vv instruction happens. See the warning and error message below.

    # UVM_WARNING /uvm/sim/../src/generic/env/cvxif_scoreboard.svh(207) @ 2000: uvm_test_top.env.scoreboard [CVXIF_SCOREBOARD] Instr "vrgather.vv v8, v0, v4 m4, tu, mu" in Prog "vrgather_32" failed: 
    # at                 2000: State Difference
    # ref.vproc_register[10]: 3f44383b37424d545e4b504900000000, dut.vproc_reg[10]: 5e4b5049000000004a505f3e4a505f3e
    # ref.vproc_register[11]: 323b3f4747434b3a302f2e3200000000, dut.vproc_reg[11]: 302f2e32000000003f44383b37424d54
    # 
    # UVM_ERROR /uvm/sim/../src/generic/env/cvxif_scoreboard.svh(223) @ 2760: uvm_test_top.env.scoreboard [CVXIF_SCOREBOARD] Program "vrgather_32" failed

You can reproduce the issue by running the `cvxif_test_direct_issue_80` test in the UVM environment.

## Findings
I found it quite strange that enabling random memory result transaction delay suddenly causes issues with the gather instruction. This is why I spent a few minutes trying to understand the cause of the issue.
I might not have found the root cause of the issue, but I will try to explain what I have found (all signals I mention below are part of the `vproc_elem` module:

The result calculated by vicuna is "basically correct", but "shifted" by two wrong elements.
In the vproc_reg[10] there are two wrong elements (bold elements below) that cause the rest of the vrgather result to be shifted by two elements.
dut.vproc_reg[10]: 5e4b5049_00000000_**4a505f3e_4a505f3e**

Now these two elements are not just some random elements, but they are equal to the last element in dut.vproc_reg[9]
dut.vproc_reg[9]: **4a505f3e**_485e5455_4d4c4a41_00000000

From what I have seen, the `pipe_out_res_valid_o` signal stays high for 3 clock cycles while the result (4a505f3e) stays the same for these three clock cycles (this explains the 3 equal elements of which two are wrong).

Now `pipe_out_res_valid_o` is equal to `result_valid_q`. And `result_valid_q` is the registered value of `result_valid_d`. `result_valid_d` is only high for 1 clock cycle (which would be correct), but `result_valid_q` is not updated because `state_res_valid_d` is low for two cycles (`state_res_valid_d` gates the update of `result_valid_q`). Below you can see line 66-74 of vproc_elem.sv which suppress the update of `result_valid_q` and thereby cause the issue with the "wrong elements".

            always_ff @(posedge clk_i) begin : vproc_elem_stage_res
                if (state_res_ready & state_res_valid_d) begin
                    { ... }
                    result_valid_q <= result_valid_d;
                end
            end

Below you can find a screenshot of QuestaSim which shows the signals I described above. The issue happens after ~1600ns.

[image]

## Possible solution
One solution might be to "not-gate" the update of `result_valid_q` with `state_res_valid_d` but as said above, if my findings are not the root cause of the issue then this solution might create other problems.

moimfeld, 2022-06-13:
While playing a bit with different tests today, I found that the same issue can happen for the `vcompress` instruction. I would assume that this issue can happen for every instruction that utilizes the elem unit.

The issue happens as soon as `state_res_valid_d` goes low.

michael-platzer, 2022-06-15:
Hi @moimfeld, thanks for reporting and thanks for the detailed analysis!

I believe this issue should also be fixed now in 917ee69. You were right that the issue can happen for all instructions executed on the ELEM unit. The problem was that the ELEM unit's output buffer only considered the `pipe_out_res_valid_o` signal, which indicates that the ELEM unit has produced a valid result, but not the `pipe_out_valid_o` signal, which must also be asserted for a valid result. This situation arises when an earlier stage of the pipeline is stalled waiting for another instruction to produce data that must be read by the ELEM unit.

moimfeld, 2022-06-15:
Hi @michael-platzer,

Thank you for the fix and the explanation! I ran the `cvxif_test_direct_issue_81` test which completed successfully, so this issue is fixed 😄.

closed 2022-06-15 by moimfeld