vproc/vicuna #89: Illegal configuration created by config.mk

moimfeld, 2022-06-27:
Hi @michael-platzer,

when using the `VPORT_POLICY=many` policy, the Makefile can generate configurations where the maximum `PIPE_W` is equal to `VREG_W`.
When one `PIPE_W` is equal to `VREG_W` then the `MAX_OP_W` of one `vproc_pipeline` is also equal to `VREG_W`. This will trigger the fatal error below:
https://github.com/vproc/vicuna/blob/d3c1cf85355ec2b6c0357f23e94f46fe8ceb9dac/rtl/vproc_pipeline.sv#L98-L102

The following lines in the config.mk cause this "issue":
https://github.com/vproc/vicuna/blob/d3c1cf85355ec2b6c0357f23e94f46fe8ceb9dac/config.mk#L37
https://github.com/vproc/vicuna/blob/d3c1cf85355ec2b6c0357f23e94f46fe8ceb9dac/config.mk#L67-L70

michael-platzer, 2022-07-04:
@moimfeld Thanks for reporting! Two units are still causing problems when the operand width equals the register width, but I am working on resolving this and will then change this fatal error to only trigger if the operand width is larger than `VREG_W`.

Aravind-kris, 2024-03-15:
> @moimfeld Thanks for reporting! Two units are still causing problems when the operand width equals the register width, but I am working on resolving this and will then change this fatal error to only trigger if the operand width is larger than `VREG_W`.

Did you solved it ?