vproc/vicuna #91: Error in Synthesizing the vicuna on Genesys2 board

Nikhil-311293, 2022-07-07:
Hello community,

I've been trying to synthesize the vicuna vector coprocessor for Kintex7 FPGA (Genesys2 Board).
Wherein I've created a **_vproc_config.sv_** package file using command -

 make -f /home/thales/Documents/Nikhil/vicuna/config.mk VPROC_CONFIG=dual VREG_W=512 VPORT_PLOICY=few TARGET_TECH=fpga

I'm trying this exercise using Vivado in GUI mode after using all the RTL source files from vicuna repo from GitHub.
I've also attached the error messages and screenshots of config pkg and vproc_core.sv for the reference purpose.

[image]

[image]

Can you please suggest if there is any solution for the above mentioned error.

michael-platzer, 2022-07-11:
Hi @Nikhil-311293,

this error indicates that the width of the VLSU's operands does not match the width of Vicuna's memory interface. In your configuration file the width of the VLSU is configured to 32 bits, hence the memory interface must also be 32 bits wide. Unfortunately Vivado does not print the full error message, thus I cannot see which value you have provided for the memory interface width.

The width of Vicuna's memory interface is controlled by the parameter `XIF_MEM_W` of the `vproc_core` module or the parameter `VMEM_W` of the `vproc_top` module. Set either of these parameters to 32 and you should no longer get that error.

Nikhil-311293, 2022-07-12:
Hi @michael-platzer

With your suggestions now I'm able to synthesize the vicuna vector coprocessor for Genesys2 board using Ibex core.
Thank you for your time.

Now I'll have to implement the design and generate the bitstream for the same.

I wanted to ask this one thing, can I port the same design for other Xilinx FPGA e.g. VCU118 ultrascale FPGA?

Nikhil-311293, 2022-07-18:
Hi @michael-platzer 

After being able to successfully synthesize the vicuna coprocessor for Xilinx's Virtex Ultrascale FPGA, now I'm trying to implement the design and generate the bitstream for it.

My ultimate aim is boot Linux on it run few simple example programs on it. The problem I'm getting now is: referring as you suggested about the _**demo**_ and **_sw_** subdirectory ---->

Do I need to put constraints for **_mem_req_o_**, **_mem_addr_o_**, **_mem_wdata_o_** etc ports which are declared in the top module **_vproc_top.sv_**??

or Do I need to initialize the MIG IP from Xilinx to serve this purpose?

As the constraints used in demo subdirectory are exclusively for UART test program, I believe.

I've attached screenshot below to illustrate this. How shall I move ahead in this particular task? Your suggestions will be extremely useful in this further progress.

[image]

Thank you in advance.