vproc/vicuna #106: fail to set VREG_W=2048

schwa1z, 2022-10-02:
Hi, @michael-platzer @stevobailey @moimfeld 
As vicuna user guide shows, the VREG_W could be set with 64/512/2048.
[image]
While when I use the following command to configure vicuna:
 ```
make -f /home/mango/my-vicuna/sim/Makefile CORE=cv32e40x PROG_PATHS=progs.txt VPROC_CONFIG=triple VREG_W=2048
```
error occured:
[image]
When I set VREG_W<=1024, no error occurs. So what's the problem?
Thanks!

michael-platzer, 2022-10-13:
Hi @schwa1z,

this is a Verilator warning that is by default promoted to an error. Verilator assumes that an 8k bit replication is wrong, but for large vector sizes it is perfectly normal to encounter such replication counts. Therefore, it should be safe to ignore that warning, which you can do by adding the `-Wno-WIDTHCONCAT` option to Verilator's invocation in the [simulation Makefile](https://github.com/vproc/vicuna/blob/main/sim/Makefile). E.g., adding that option to line 107 after the other options that silence various warnings should resolve this problem:

https://github.com/vproc/vicuna/blob/bda35112a983e9adb4ea8413b0afc2f0d91e7563/sim/Makefile#L107

schwa1z, 2022-10-13:
Thanks! It's clear!

closed 2022-10-13 by schwa1z